Schildkroet/GRBL-Advanced #1: Laser mode

dplund, 2018-04-21:
I got it to run on my nucleo board and configured it for laser engraving, thanks for sharing this!

Is there a way to turn on laser mode for this version? The settings menu doesn't provide an option for this, is it commented out or just not included?

Programs seem to run very slow, like, 1 block per second, though it runs full speed when executing each block, but sits for about a second after each one.

It seems it will only connect at 115200 baud. This probably won't be a limitation, but for grayscale rastering it can push the limits of the bandwidth, I am running my STM32F103 at 230400 baud and the buffer occasionally does dip below half.

Thanks in advance for your consideration and help.

Schildkroet, 2018-04-21:
Since this project was intended for my CNC mill, i didn't care about laser mode. I even might have removed some code for laser  mode...? GRBL settings etc stayed the same, so you should check grbl1.1 for laser mode and how to configure it. If settings don't work, you have to look into source code for changes.

For the speed problems:
1. Make sure you set the Target in EmBitz to 'Release'. It uses optimizations and programs are usually twice as fast as Debug Target.
2. I tried to be as much compatible with grbl as possible, so i set the baudrate to 115200. But you can change it to 230400 (or even more) in grbl/Config.h. 
In HAL/USART/USART_FIFO.h you can also increase the serial buffer. The mcu has plenty of RAM left. But this works only, if you tell your streaming software, about the extended buffer.
In grbl/Config.h you can also increase sizes for planner etc.

mfg

closed 2018-04-23 by Schildkroet

dplund, 2018-04-23:
Thanks for looking into this. I'll work on the laser mode and let you know
if I make any progress I think might be helpful.

Closed #1 <https://github.com/Schildkroet/GRBL-Advanced/issues/1>.