Schildkroet/GRBL-Advanced #3: Slow processing

darklarcel, 2018-05-24:
Hello, thanks for the project you are doing, I just bought the board STM32F411RE NUCLEO and tested with my cnc Co2 laser, previously it worked with Arduino Uno but it was too slow because of the speed of Arduino processing. I decided to jump to the platform of STM32 with your port of grbl 1.1, I expected to get more speed by 100Mhz compared to 16Mhz of Arduino but it has not been like that, the speed at which it starts to block is F: 6000 almost the same as Arduino (in the engraving of pictures with dithering, for images with no load as logos or text no problem, can reach speeds of F: 20000, that's why I think it's processing speed problem), I checked the code of the STM32 project and the core clock is the Right. I do not know what I can be doing wrong or it depends on more factors (I have also reduced the number of microsteps to 1/8). My settings are:

$0=15 (Step pulse time)
$1=20 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=10000.000 (Homing locate feed rate)
$25=10000.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=20.000 (X-axis travel resolution)
$101=20.000 (Y-axis travel resolution)
$102=1203.000 (Z-axis travel resolution)
$110=8000.000 (X-axis maximum rate)
$111=10000.000 (Y-axis maximum rate)
$112=1200.000 (Z-axis maximum rate)
$120=700.000 (X-axis acceleration)
$121=6000.000 (Y-axis acceleration)
$122=80.000 (Z-axis acceleration)
$130=70.000 (X-axis maximum travel)
$131=300.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Thanks in advance!

Schildkroet, 2018-05-24:
I think this is mainly a problem of USART baudrate. Set a higher baudrate and test if it works better. Also see this post: https://github.com/Schildkroet/GRBL-Advanced/issues/1#issuecomment-383282917

mfg

darklarcel, 2018-05-24:
You're right! I'll try it today and I'll tell you the results,
Thank you for responding so fast!

darklarcel, 2018-05-24:
Said and done, first increase the Baudrate to 230400, but there was no improvement, even so I have kept it said value, I have finally increased the BLOCK_BUFFER_SIZE from 36 by default to 256, I have made several tests and now it is able to maintain a speed of F: 20000 with large data load (photographs).
F: 20000 is the maximum for my machine, I'm sure it can reach something else without problem.
Thanks for the help and my congratulations for all the work!

closed 2018-05-24 by darklarcel

darklarcel, 2018-05-25:
Here are the tests!:

STM32F411RE FEEDRATE(WITH LOAD) TESTS

BAUDRATE 115200

1-TEST
BAUDRATE 				                        115200
BLOCK_BUFFER_SIZE			                36(BY DEFAULT)
MAX STABLE FEEDRATE(WITH BIG LOAD)	7500

2-TEST
BAUDRATE 				115200
BLOCK_BUFFER_SIZE			128
MAX STABLE FEEDRATE(WITH BIG LOAD)	9000

3-TEST
BAUDRATE 				                        115200
BLOCK_BUFFER_SIZE			                256
MAX STABLE FEEDRATE(WITH BIG LOAD)	9800	

BAUDRATE 230400

1-TEST
BAUDRATE 					                230400
BLOCK_BUFFER_SIZE			                36(BY DEFAULT)
MAX STABLE FEEDRATE(WITH BIG LOAD)	12000

2-TEST
BAUDRATE 				                        230400
BLOCK_BUFFER_SIZE			                64
MAX STABLE FEEDRATE(WITH BIG LOAD)	13800

3-TEST
BAUDRATE 				                        230400
BLOCK_BUFFER_SIZE			                96
MAX STABLE FEEDRATE(WITH BIG LOAD)	15000

4-TEST
BAUDRATE 				                        230400
BLOCK_BUFFER_SIZE		                	128
MAX STABLE FEEDRATE(WITH BIG LOAD)	16000

5-TEST
BAUDRATE 	                           			230400
BLOCK_BUFFER_SIZE		                	166
MAX STABLE FEEDRATE(WITH BIG LOAD)	16800

6-TEST
BAUDRATE                             				230400
BLOCK_BUFFER_SIZE	                    		200
MAX STABLE FEEDRATE(WITH BIG LOAD)	17000

7-TEST
BAUDRATE                        				230400
BLOCK_BUFFER_SIZE                			256
MAX STABLE FEEDRATE(WITH BIG LOAD)	19000

8-TEST
BAUDRATE 	                            			230400
BLOCK_BUFFER_SIZE	                 		>256
MAX STABLE FEEDRATE(WITH BIG LOAD)	----- no improvements
--------------------------------------------------------------------------------
the most limiting factor is the baudrate, but the BLOCK_BUFFER_SIZE also, I think it could continue to improve by increasing the baud rate.

My STM32 board is like the left picture, whitout the crystal. Is it a problem?

Schildkroet, 2018-05-25:
Thanks for your tests! Thats good data.

No problem with crystal. It's just that i use a external 8 MHz crystal (which is soldered by myself, since these boards usually come without one) and tested only this way. 
I included a fallback to the internal clock if there is no external detected, but couldn't test it. Thanks to you i know now that it works.

mfg

darklarcel, 2018-05-29:
Good news !, I've been busy these days, but I managed to increase the baud rate to 460800, the difference is impressive, now it can reach speeds of F: 27000 with load (I also had to modify the LaserGRBL code), so much speed is unnecessary. I work with GT2 belt and 16 tooth pulleys but now thanks to so much speed I can increase the microsteps and maintain a good speed! One more question, could GRBL Advanced be used on the STM32F444RE board? her outputs are equal, but its speed reaches 180MHz, thanks!