Schildkroet/GRBL-Advanced #4: Bad acceleration / deceleration and noise

darklarcel, 2018-06-09:
Hello again!,
I've been testing your port again (I've been very busy these days),
After solving the problem of the low speed of processing, I was performing tests with the laser cnc and when exceeding speeds of F: 10000 appear blockages and lost steps when accelerating and decelerating, I use an X raster as a technique of engraving, also the noise has increased quite a lot compared to arduino.
My settings are:

$0=15 (Step pulse time)
$1=20 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=10000.000 (Homing locate feed rate)
$25=10000.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=20.000 (X-axis travel resolution)
$101=20.000 (Y-axis travel resolution)
$102=1203.000 (Z-axis travel resolution)
$110=8000.000 (X-axis maximum rate)
$111=10000.000 (Y-axis maximum rate)
$112=1200.000 (Z-axis maximum rate)
$120=700.000 (X-axis acceleration)
$121=6000.000 (Y-axis acceleration)
$122=80.000 (Z-axis acceleration)
$130=70.000 (X-axis maximum travel)
$131=300.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Use belts gt2 and pulleys of 16 teeth, the drivers are set to 1/4

Arduino worked perfectly with the same configuration (but was not able to process quickly with big data load).
I have been alternating between Arduino and STM32 NUCLEO to compare and the difference is remarkable, the arduino raster movements are smooth, silent and precise, while with STM32 Nucleo the noise increases a lot and only when accelerating and decelerating always in the same point (just before starting / finishing) he suffers a sudden braking and then continues.
I have modified config.h doing several tests with ACCELERATION_TICKS_PER_SECOND and other parameters but without success, I checked and matched the configuration with that of my arduino (by default of grbl) but I have not had any luck either.
These problems only happen when a speed of F: 10000 is exceeded, neither decreasing the acceleration the problem disappears. I think it's a problem of frequency (because of the noise) or with the acceleration algorithm.
I hope you can help me, Thanks!

Schildkroet, 2018-06-09:
I will look into it the next days. Maybe it's a problem with overflowing variables etc.

But are you sure your settings are correct? Your highest feedrate (Y-axis) is 10000, so all values above should be ignored.

mfg

darklarcel, 2018-06-10:
Solved !, in stepper.c, the levels to calculate the acceleration, (AMASS_LEVEL) were with a value of 24000000HZ, I have changed it to the value of the variable F_CPU (96000000UL) and now works perfectly, without noise, or blockages when accelerating . this is the changed code:

#define MAX_AMASS_LEVEL		3
// AMASS_LEVEL0: Normal operation. No AMASS. No upper cutoff frequency. Starts at LEVEL1 cutoff frequency.
#define AMASS_LEVEL1			**(F_CPU/8000)** // Over-drives ISR (x2). Defined as F_CPU/(Cutoff frequency in Hz)
#define AMASS_LEVEL2			**(F_CPU/4000)** // Over-drives ISR (x4)
#define AMASS_LEVEL3			**(F_CPU/2000)** // Over-drives ISR (x8)

#if MAX_AMASS_LEVEL <= 0
  error "AMASS must have 1 or more levels to operate correctly."
#endif

#ifdef MAX_STEP_RATE_HZ
    #define STEP_TIMER_MIN          **(uint16_t)(F_CPU** / MAX_STEP_RATE_HZ)
#else
    #define STEP_TIMER_MIN          (uint16_t)(50000)
#endif

Schildkroet, 2018-06-10:
OK, i found the error. I forgot to to enable register preload for the overflow event, which caused the timer in some cases to count up to its maximum register value instead of the set value.
Please see latest commit for changes and test if it works for you
https://github.com/Schildkroet/GRBL-Advanced/commit/bef526b2a380d446926c3e40c7acec0af8f00cc5

mfg

closed 2018-06-13 by Schildkroet

darklarcel, 2018-06-13:
I've tried it and it works perfectly! (It is strange that it was solved with the changes that I made), good job! I am very happy, your work could be the replacement of arduino grbl for cnc high speed laser!

Schildkroet, 2018-06-13:
Due to the error, the stepper interrupt could be delayed up to 2,7ms, which caused the noise. But it was only visible with high acceleration/speed.
By setting the frequency to 96MHz (instead of the 24MHz), you slowed down everything by 4. So if you moved with a feedrate of 10000, you actually moved with 2500, but grbl still thought it was running with 10000.

darklarcel, 2018-06-13:
Here is a test of speed and precision:
F: 25000
Resolution: 20 Lines / mm
Size: 280mmx240mm
Material: Granite
Engraving time: 36m
[image]
the text is in Spanish
Works perfect!