Schildkroet/GRBL-Advanced #5: 4 axis

enricoviale77, 2018-06-23:
is there a possibility to implement a 4 axis?

Schildkroet, 2018-06-24:
Yes, that would be possible. But this project will only receive bugfixes since there is a better ARM version coming up.

closed 2018-06-24 by Schildkroet

hdo, 2018-07-02:
Hey Schildkroet, where will the new ARM version be published? On GNEA?

Schildkroet, 2018-07-02:
Yes, according to the description, it will be published at https://github.com/gnea/gnea
But i will fork it and add STM32 support

hdo, 2018-07-02:
Ok great, thanks for feedback and Schöne Grüße ;-)

enricoviale77, 2018-07-02:
very well! I can not wait! but does a forum exist? I have a lot of questions but I do not want to disturb here!

Schildkroet, 2018-07-02:
Theres a discussion here: https://github.com/gnea/grbl/issues/67
Thats all i know