Schildkroet/GRBL-Advanced #6: invert coolant flood and spindle enable pin

alk1913, 2018-07-12:
hello,thanks for your great work,i begin to upgrade my cnc form arduino uno to stm32f411re,using your project.but i can't get work the invert coolant flood,i uncomment the define in config.h but not work.
and the spindle enable pin on D14 always remain in 0v.
thanks for your help!!

Schildkroet, 2018-07-12:
It seems i forgot to add the config header. I added them now but hadn't time to test. See if it works now

alk1913, 2018-07-14:
hello ,thanks for your work,i tested ,the invert coolant flood pin work fine,but the invert spindle enable pin doesn't work,the spindle enable pin D14 show always 0v.may be i doing something wrong?.
thanks for your help.

Schildkroet, 2018-07-16:
Update to latest commit and enable **#define SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED** in **Config.h**.

mfg

closed 2018-07-16 by Schildkroet