Schildkroet/GRBL-Advanced #7: Question grbl-advanced on i.mx8m

eric-schleicher, 2018-09-26:
More of a question..  Would it me possible to run GRBL-Advanced on one of these?

https://emcraft.com/products/868

It's powered by an NXP i.mx8 which has an M4 Arm unit.  if so would you be interested in receiving a unit to try?

Schildkroet, 2018-09-26:
Hello,
this project will most likely only receive bug fixes, because soon there will be a new version of grbl which is specifically designed for ARM including a HAL and various improvements. I plan to port the upcoming version to an stm32f446 and also integrate a better communication protocol/interface. Therefore i don't recommend to port this version, which will be out-dated soon.

Even though i worked with an imx6 SoM running linux (yocto project) on it, i have no experience in using a coprocessor.

To answer your question:
yes, running grbl on such a board shouldn't be a big problem, since it needs very litlle resources and you just have to use some GPIOs to control external hardware. But to to so, you need good skills in embedded linux which i am lacking. It would take me probably weeks or months just to bring the system in state i can start working with.

eric-schleicher, 2018-09-27:
Thanks, I'll stay tunes on a version for stm32f446

closed 2018-09-27 by eric-schleicher