Schildkroet/GRBL-Advanced #8: How to build for a F401RE board

jappiemike, 2018-10-26:
Hello, I'm a beginner when it comes to compilers etc and need your help. I managed to get Embitz running on my mac via virtual box and did a test build of your port without any problems (an unbelievable first for me) how do I change the build settings to target a F401RE board as I have a couple of these lying around.
Thanks
Mike

Schildkroet, 2018-10-27:
The F401 is also in the Access Line like the F411, so it shouldn't be that hard.....but
- Only 84 MHz CPU, so you have to edit the clock tree
- Linker file for F401

This changes should be already enough, since the other hardware features are basically the same. But i don't own a F401 to test and for the next version i will target a F446.

jappiemike, 2018-10-27:
Sorry you've lost me already. I did find a .ld file for the f401re which looks the same as your 411 file apart from the memory sizes. How do I change them in your project. Which file is the clock tree in?
Cheers 
Mike

Schildkroet, 2018-10-27:
You can just change the memory sizes in the current linker scipt or go to Project->Build Options->"Device"-Tab and change the linker script there.

The clock settings are in HAL/STM32/system_stm32f4xx.c
You can use CubeMX to calculate the correct values for your clock settings

jappiemike, 2018-10-27:
It looks like the clock file is the same for both chips and I've not made any changes. I've changed the linker script to use the 401 file and also left the stack and heap sizes the same, and it built with no errors. I'll program a board and see how it goes.
Thanks for the help.
Cheers
Mike

jappiemike, 2018-10-28:
Well I reached the limit of my knowledge and took the easy way out, my 411re board will be here next week. Thanks for the help.
Cheers
Mike

closed 2018-10-29 by Schildkroet

jappiemike, 2018-10-31:
Sorry,I'm still stuck. I got a 411 board today and moved the .hex file to the board but it didn't run. I updated the firmware and I can load bin files to the board and they all run. Is there something I'm dong wrong. I've tried this using both a mac and Virtual box without success.
Cheers
Mike

Schildkroet, 2018-10-31:
The mass storage on the board needs a .BIN file to flash.
You either convert the .hex to .bin using objcopy or you use another tool like st-util to flash the .hex

jappiemike, 2018-10-31:
Success, all I had to do was spend 6 hours working out that I needed admin. mode on my virtual box, then download all the files (and a LOT I didn't need) then once st-util was installed along with all the drivers it was a piece of cake ( a letdown really, I was expecting a bigger battle). I used the serial monitor in the arduino ide to get the feedback from the program. Thanks for the tips.
Cheers
Mike