Schildkroet/GRBL-Advanced #9: Program continues after alarm

jappiemike, 2018-11-07:
Hello, I've noticed that if the machine has an alarm (exceeding soft limits as an example), once the error is cleared the remainder of the travel is completed. My uno board doesn't do this, in fact once I noticed the problem I swapped them straight away. I'm using a screw shield so there was no swapping of wires between swaps. I've used the same configuration file for both boards and I'm using USG. 
Cheers
Mike

Schildkroet, 2018-11-07:
Serial buffer was not cleared after reset. Should now work as expected
Commit 06870575809692686c86e5342c8802127b91984c

closed 2018-11-07 by Schildkroet

jappiemike, 2018-11-16:
Hello, the problem still remains, most noticeable after homing the machine. Also tonight when using the nucleo board with UGSender it crashed after about 10 lines and I couldn't get it to communicate again.
Any ideas?
Cheers
Mike

Schildkroet, 2018-11-16:
Please post your build info, your settings and the program it crashes (you can attach files here), so i can test it.

jappiemike, 2018-11-18:
Hi, here are my grill settings 
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 1000.000    (Homing search seek rate, mm/min)
$26 = 25    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 1000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 118.000    (X-axis travel resolution, step/mm)
$101 = 118.000    (Y-axis travel resolution, step/mm)
$102 = 1280.000    (Z-axis travel resolution, step/mm)
$110 = 1500.000    (X-axis maximum rate, mm/min)
$111 = 1500.000    (Y-axis maximum rate, mm/min)
$112 = 600.000    (Z-axis maximum rate, mm/min)
$120 = 20.000    (X-axis acceleration, mm/sec^2)
$121 = 20.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 580.000    (X-axis maximum travel, millimeters)
$131 = 640.000    (Y-axis maximum travel, millimeters)
$132 = 130.000    (Z-axis maximum travel, millimeters)
$140 = 0.050	{X-axis backlash, millimeters)
$141 = 0.050	{Y-axis backlash, millimeters)
$142 = 0.050	{Z-axis backlash, millimeters)

AS I stated earlier these work fine with the uno board in place but not with the nucleo board.  I noticed there were incorrect motion of the Z and Y axis when jogging right after homing the machine but the motion of the 3 axis seemed ok after the first movement. 
Here is the test program I was running
 it’s nothing special I just wanted to see how the nucleo was performing, it runs fine with the uno.
Thank’s for the help.
Cheers
Mike

Schildkroet, 2018-11-18:
I see you are using the backlash branch. The first version had a serious issue and was causing wrong behaviour in complex motions. Make sure you are using the latest version and check again if it works.