Schildkroet/GRBL-Advanced #13: Probe issues

spothus, 2019-01-09:
Hello and happy new year,
The probe doesn't stop on contact (it was working with the arduino uno and a CNC shield)
I checked your code and it looks like the wiring pin should be the same.
Any advice ?
Thanks

Schildkroet, 2019-01-09:
Usually the probe should trigger when pulled to ground. But i made a few changes in the gpio config recently, therefore it might be possible that something went wrong. I will look into it in the next few days.

spothus, 2019-01-09:
Thanks, perfect.
In gpio.c it is dedicated to Analog 5 of the Arduino and PC0 of the STM32 witch looks correct.
But it looks like it is mixing with the safety door pin because as soon as I plugged my probe I got the “door open” message. For that I added the conditional compilation excluding this test if the ENABLE_SAFETY_DOOR_PIN (or something) is not defined (it is missing in system.c).

Schildkroet, 2019-01-12:
I tested the probe feature with following cmd: `G21G91G38.2Z-30F100`
It stopped after being tied to ground like it should. There must be an error with your setup.

closed 2019-01-12 by Schildkroet