Schildkroet/GRBL-Advanced #16: Pin assignments

PlanBV2, 2019-05-13:
Hi Schildkroet,

thanks for sharing this project!

I'm planning to create a shild for the Nucleo board with RC filters for the limit switches, EEPROM, etc... For this reason, I have looked through the pin assignment, and I found that the pin description in the included picture (nucleof411re_pinout.png) is not up to date. Also, the pins for lim_Y and lim_Z are both the same (Pin_6). Are you planning to change the pin assignment in the near future? Then it wouldn't make sense to design a PCB for the actual pin assignment.

best Andi

Schildkroet, 2019-05-13:
What exactly is wrong in the pinout image? LimY and LimZ are on different Ports, so nothing wrong here.
The only pin which will change is the door pin, since it interferes with the external eeprom.

PlanBV2, 2019-05-13:
Oh sorry, I was wrong on the Ports... Somehow I thought  Lim_Y is also assigned to port A. My mistake. Thanks for clarifying :)

closed 2019-05-13 by PlanBV2