Schildkroet/GRBL-Advanced #17: 3D Printer Support

Geramy, 2019-05-15:
Hi, I was wondering if this platform is capable of running a 3D Printer in the sense of pins and hardware. I would like to help add support if we can support up to 2 extruders at least, that being said it would need to support not only 2 extruders but uart and/or spi motor drivers too.
Some icing on the cake would to be able to also have SPI support for 2 MAX6675 chips.
Of course I would be able to write a huge amount of these drivers needed for this, but I was hoping you could provide insight to if this is the right platform for such things.

mikemaikaefer, 2019-05-16:
Extruders and other control parts are out of the scope of GRBL. It can run the XY(Z) motion and do the material feed but for all the other things you definitely need a separate software which can do these things.

Geramy, 2019-05-16:
Gotcha, I been trying to find a really good base to start with, i would like to make a firmware that works better then smoothieware or Merlin dont get me wrong there great but there are some weird bugs with both and some features not in them and 180mhz processor would make a really good controller.

mikemaikaefer, 2019-05-16:
When you are looking for a really good solution, you should use separate motion stage controllers (like GRBL) for X, Y and Z, separate motion controllers for extruders/material and a completely separate control software running on an own computer system (may be a PC, may be an embedded system). But an all-in-once-firmware on a weak device will always suffer from the lack of flexibility and resources. Examples for such a control software can be found e.g. at https://replisls3d.com or with google.

Schildkroet, 2019-05-16:
Hello,
in terms of hardware it should be no problem, but grbl (advanced) is just for cnc-milling etc. It has no 3d printing capability. You would have to implement dozens of new G/M-Codes.
I don't intend in doing so, but you are free to fork and modify this code as you wish.

closed 2019-05-16 by Schildkroet