Schildkroet/GRBL-Advanced #19: Build Question(STM32F429ZI)

Lim-Seunghwan, 2019-05-24:
Hi Schildkroet!
Thank you for sharing your wonderful  project. 

Before, I tested GRBL with Arduino-Mega and successfully upload it. 
And now I want to use GRBL on STM32 Board and found this project.

I tried to upload GRBL on Nucleo Board144(STM32F429ZI)
I followed your suggestion guide(https://github.com/Schildkroet/GRBL-Advanced/issues/8). But finally failed ... :( 
 I think there are some missing point. Could you give advice about it?

Here is what I changed for STM32F429ZI
(1) Add linker file 
- stm32f429zi_flash.ld
- stm32f429zi_sram.ld

(2) Change Compiler Setting 
[image]
[image]

(3) Change Clock Tree 
[image]

And I make bin file and put in to nucleo board by drag and drop on File Explorer.
But I can`t get response from board. Is there are more things to change?

Best Regards 
Seunghwan

Schildkroet, 2019-05-24:
Which clock source are you using? Did you configure the HSE/HSI value accordingly?

Lim-Seunghwan, 2019-05-27:
Hi Schildkroet

I made new Embitz project for STM32f429ZI and refered clock source.
For clock tree, I just change PLL_M, PLL_N, PLL_P, PLL_Q, SystemCoreClock value in "STM32/system_stm32f4xx.c"

Is there more values to change clock tree?

Schildkroet, 2019-05-27:
Since you set PLL_M to 25, i assume you are using a 25MHz ext crystal? You also have to set HSE_VALUE in https://github.com/Schildkroet/GRBL-Advanced/blob/HAL/STM32/stm32f4xx.h to 25000000.
In https://github.com/Schildkroet/GRBL-Advanced/blob/software/grbl/util.h you have to set F_CPU and F_TIMER_STEPPER for correct stepper timing

Lim-Seunghwan, 2019-05-27:
Hi Schildkroet
Thank you for your reply.

It is first time for me to use STM32 board. 
So I am not familiar with clock configuration, therefore I am trying to build GRBL with default configuration as your project. (After practice, I will try to calculate clock for my purpose.)

Currently, my goal is receive message from board like picture below. (Picture is result with Arduino-Mega)
[image]

I restart project and followed with your clock(96000000).
And I use STM32CubeMX to calculate clock data for my board.
[image]

I changed with upper clock setting.
PLL_M : 8
PLL_N : 96
PLL_P : 2
PLL_Q : 4

But still I can`t receive message from board.... 
Is there are more values to change when I change PLL values?

Schildkroet, 2019-05-27:
If you didn't modify your board, it should be clocked by the st-link with 8 MHz. So it should run with 96 MHz without any modification of the clock tree. Did you try it?

I have the same board and will test it later today.

Schildkroet, 2019-05-27:
Hello,
you have to reconfigure USART interface. On Nucleo64 the default for VCP is USART2, but on Nucleo144 it is USART3. Then you will see ouput in terminal.

This clock config should work:
[image]

Lim-Seunghwan, 2019-05-29:
Hello Schildkroet

Thank you for your kind answer:) 
I will try it with your suggestion.

closed 2019-06-15 by Schildkroet