Schildkroet/GRBL-Advanced #21: Hold Status Issue (STM32F429ZI)

Lim-Seunghwan, 2019-07-04:
Hello Schildkroet!

With your advice, I uploaded GRBL on my Nucleo STMF429ZI.(USART2 -> USART3 was a solution)
So I can get response from Candle2(also your version) like below.

[image]

And now I have one problem in board status. 
With STMF411RE, after press 'reset' button, status change to 'Idle'.
But with STMF429ZI, it does not change to 'Idle', stay in 'Hold' status.

Is there more things to change for use STMF429ZI?

Best Regards,
Seunghwan.

Schildkroet, 2019-08-01:
Hello,
it seems the 'Feed Hold Pin' is triggered. Maybe there is some external peripheral on this pin with the F429.

Lim-Seunghwan, 2019-08-14:
Thank you for answer. I solve this problem by change the 'Feed Hold Pin'.

closed 2019-08-14 by Schildkroet