Schildkroet/GRBL-Advanced #22: When USART_NUM = 1

klchin66, 2019-08-05:
Hi, 

I had try to your code using Atollic with  stm32f407zgt and stm32f407igt,
(both china both), I got 2 different result. I Just testing the code without 
connecting to any hardware - stepper, switch and etc. 

a)  407ZGT it work  -- luckily it.

b)  407IGT it work but it stopped with click the Unlock from the Candle2.0
    - when the board start, on candle  it got ERROR:Unknown or ERROR:9
     - when candle send  $X and received OK
     - following candle send the spindle speed S10000 but not received OK
       and stop here.

    what could be the problem? 

BTW, I suggest when USART_NUM = 1,  code under the stm32f4xx_it.c, 
added this checking, it can work without any changes ... just like when 
run it, wonder why the candle not showing the real time status  and position.

a) 
void USART1_IRQHandler(void)
{
	if(USART_GetITStatus(USART1, USART_IT_RXNE) != RESET) {
		/* Read one byte from the receive data register */
		unsigned char c = (USART_ReceiveData(USART1) & 0xFF);

		//  Write character to buffer
    if ( 1 == USART_NUM )         //  Added this checking
      ProcessReceive( c );
    else
      FifoUsart_Insert(USART1_NUM, USART_DIR_RX, c);
	}

b) 
void ProcessReceive(char c)
{
...
...
		}
		else {
			// Write character to buffer
		  if ( 1 == USART_NUM )       //  Added this
                    FifoUsart_Insert(USART1_NUM, USART_DIR_RX, c);     //  Added this checking
		  else
		    FifoUsart_Insert(USART2_NUM, USART_DIR_RX, c);
		}
	}
}

TQ.

Schildkroet, 2019-08-05:
Hello,
Error 9 indicates, that you sent gcode while grbl is not in idle. It might be possible, that your second board performs a reset after unlock (due to a hard_fault).

USART_NUM was meant to easily switch on/off usart interface and save RAM if only 1 interface is used.

klchin66, 2019-08-06:
Thx for the answer.

Still struggle to find the cause, it hit problem when it receive S10000 on the 1st 's' character.  
USART1 not receive anymore interrupt when using debugger.  It was somewhere in the 
while loop I think , because it did not stop at while(Getc(&c) == 0) loop under 
Protocol_MainLoop. 

USART_NUM do understand.  The reason mentioned, because the if using 1 port (USART1) 
and the ProcessReceive not called without change USART.c..

Is that right?, You implement minimum 2 ports, mainly 1 for Protocol_MainLoop.  
and 1 for command - real time ProcessReceive or update positions/status/override.

TQVM.

klchin66, 2019-08-06:
I found the cause, it was cause by SysTick_Handler  CONTROL_FEED_HOLD_BIT triggered.
Because do not aware that PORTA PIN 1 used in this file - system.c. 

TQVM.

closed 2019-08-07 by klchin66