Schildkroet/GRBL-Advanced #23: Troubles with connecting to GRBL

czapeczek, 2019-09-26:
I am having regular problems with connecting to GRBL. After I deploy the binary, everything is fine and I can use GRBL. But after turning it off and then on - I cannot connect. There is no response on Serial port from the device. 

I really doubt that this is a bug but no clue how to track this. Did anyone encountered such problem?

Schildkroet, 2019-09-26:
How do you flash the binary?

czapeczek, 2019-09-26:
I build it with EmBitz, no warnings
Flashing by copying the bin file to the NUCLEO drive which appears after you connect it to the USB

I tried to see how does that work and it's really weird. 
I am not able to simply send a new build and get thing working. 
I need to upload some different code first. Like simple led blinking code. Then I can send GRBL Advanced and only after that I can connect

czapeczek, 2019-09-26:
Embitz 1.11 to be more precise

Schildkroet, 2019-09-27:
When building with Embitz, make sure to select 'Release' as target. There will be a HEX file in the output folder.
Flash the HEX with ST-Link Utility or STM32CubeProg.

czapeczek, 2019-09-28:
Nothing changes. I might buy another nucleo and see if there's a difference. But sounds really odd. 
Thanks for your help. If anything comes up to your mind I'll appreciate help

Cheers
Filip

czapeczek, 2019-09-28:
@Schildkroet I upgraded ST link firmware and it gives me some hopes. I will let you know in 1-2 days if it's stable

closed 2019-09-28 by czapeczek