Schildkroet/GRBL-Advanced #24: LAN connectivity troubleshooting

czapeczek, 2019-10-03:
Hi, 

I connected W5500 to the setup. It works fine unless I attach GRBL board to NUCLEO. After that I have a timeout on connection. After removing the shield I have no problems at all. 

Did you discover similar issues? Are you able to point me where to troubleshoot?

Thanks!
Filip

Schildkroet, 2019-10-04:
What board did you connect and how?
W5500 draws a lot of power. Try an external power supply for W5500.

czapeczek, 2019-10-04:
I connected simple W5500 board
[link to te polish auction site](https://allegro.pl/oferta/modul-sieciowy-ethernet-w5500-arduino-8251446649)
I tried separate power supply for w5500 - still same result and yes - it takes about 0.5A

the borad is connected to nucleo accordingly to scheme given in the repository
3.3V, SPI3 

still - when the grbl board is off nucleo it works (i get ping) 
when I connect grbl board there's no response. 
I might try SPI3

Schildkroet, 2019-10-04:
Sorry, i meant which grbl board are you using and how is it connected? Usually it should not be a problem, since the SPI for W5500 is not related to arduino pinout.

Are you sure the nucleo itself (grbl) is running when you connect the grbl board? This boards are usually designed for 5V. So it might cause problems.

czapeczek, 2019-10-04:
Interesting. I even didn't think before about the voltage difference. However I've successfuly ran my CNC machine with the GBRL board ver. 3.00 connected to nucleo via standard arduino pinout. I made several laser cuts with this configuration
Maybe it work cause I use DRV8825 drivers. 

Do you think the voltage difference can cause some interference?

Schildkroet, 2019-10-04:
DRV8825 can run with 3,3 and 5V. So they are fine.
But you said it runs fine without the grbl board, so there must be a problem with it. Are you sure, everything is connected right? What happens if you connect the grbl board after you ping it already?

czapeczek, 2019-10-04:
everything off, no board, turn on = ping
everything off, grbl board in, turn on = no ping
everything on with no grbl board = ping. after i put in first pins there is no ping anymore. I'll  try same but without grbl power supply

czapeczek, 2019-10-04:
Yes. Something's going wrong after I plug in GRBL power supply

czapeczek, 2019-10-04:
Just to add more curiosity:
When I turn on nucleo without grbl i Have ping
When I connect grbl without PS still there is ping
When I turn the power on, no ping anymore
After I turn off grbl PS and reset nucleo, no ping! :D
I will not have ping until I take out grbl board turn on nucleo without grbl board 

I
don't 
get 
it

⛱

Schildkroet, 2019-10-04:
Did you also connect PA_15 (W5500 Reset)?
There seems to be a short circuit in your grbl board. The W5500 has its own IP Stack and runs/can be pinged (after configuration) as long as power is supplied

czapeczek, 2019-10-04:
Yes. The reset pin was connected. No changes after pulling it off

czapeczek, 2019-10-04:
I will try to measure differences in voltage on ping after supllying grbl with power. Will let you know - thanks a lot for the guidance.

czapeczek, 2019-10-04:
I made it work. 
I don't really know what happened. 
What I did is I pulled the laser PWM cable off the grbl board. 
Still with troubles but now I think it works. 

I am not closing this issue just in case I cannot figure out what's happening.

czapeczek, 2019-10-04:
Still I am having some issues - after I run some GCODE file, after several instructions I get:
[MSG:Reset to continue]

Soft Reset doesn't work then
If I disconnect and try to reconnect I am getting 
Ethernet error!Connection refused

Then I need to make a hard reset - turn it off and on again but the pattern remains. 

I'll try to buy another GRBL board or if you have any other ideas what can it be - just shout

Schildkroet, 2019-10-05:
Since this problem seems to be due to your grbl board (which i don't use), i can't help you that much. You have to try yourself.

closed 2019-10-05 by Schildkroet

czapeczek, 2019-10-05:
Yes. Thanks A lot for your time and help! What do you use, btw?

Schildkroet, 2019-10-05:
I designed my own pcb