Schildkroet/GRBL-Advanced #27: Spindle not spinning at desired speed

doctorpartlow, 2020-02-24:
I'm having an issue when running the spindle via pwm. I essentially copied the settings I had on grbl-mega into this  (grbl-advanced is soo much smoother, thanks the work on it)  the grbl mega iteration spun up fine at 24000rpm but without looking into the code on grbl-advanced, when issuing a M3 S24000 the spindle only spins at 18 18388rpm. I'm not sure if this is something I did or not. let me know and I will provide any extra information you need. Thanks

Schildkroet, 2020-02-24:
Hello,
1. Pls Post your Dettingen: $$
2. If you can measure the voltage of the Spindle Pin, Set ist to max rpm. It should be 3,3V.
3. Are you aware that ARM MCUs run only with 3,3V? If your spindle ist designed for 5V AVR, you need a level shifter

doctorpartlow, 2020-02-25:
that makes perfect sense then. so would a logic level converter from 3.3 to 5v remedy the issue?

Schildkroet, 2020-02-25:
Well, i don't know your setup, but this seems like the most valid reason to me.

closed 2020-03-04 by Schildkroet

doctorpartlow, 2020-03-23:
Thank you for porting this to the nucleo board. it runs so much smoother than a mega that was previously used. sorry I didn't have time to report back until yesterday to be able to attempt some changes for this to work. connecting a logic level converter didn't work since the spindle is variable voltage input and not really pwm. so I tried a few things and got it to spin back up to 24000 with a boost converter. though the numbers are completely off. e.g. when setting an m3 s15000 the spindle goes to 24000 when at 14500 it spins at around 19500. there seems to be an exponential curve when using a boost converter for this purpose. I realize it's not the correct method to use, but it's something I had laying around and attempted it. it works for now and can will be able to determine the correct speed by playing around with the numbers until I find the right rpm the vfd indicates. so sorry about this. it isn't an issue with the code. my settings are below, even though they're not really necessary.. one thing I did notice when coming to a stop and turn the spindle slows down for a fraction of a second. I'm sure putting a capacitor inline with the output from the spindle pwm pin would correct this but for whatever reason something is sending either a slower signal to the vfd or the vfd is reading it weird as the rpm's drop ~2000. it seems to only happen after a sharp angle 

$0 = 5    (Step pulse time, microseconds)
$1 = 0    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 2    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.050    (Junction deviation, millimeters)
$12 = 0.010    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 196    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 24000    (Maximum spindle speed, RPM)
$31 = 7200    (Minimum spindle speed, RPM) <-- this was also set to 0
$32 = 0    (Laser-mode enable, boolean)
$100 = 161.448    (X-axis travel resolution, step/mm)
$101 = 20.000    (Y-axis travel resolution, step/mm)
$102 = 201.800    (Z-axis travel resolution, step/mm)
$103 = 10.000   
$104 = 10.000   
$110 = 4000.000    (X-axis maximum rate, mm/min)
$111 = 8000.000    (Y-axis maximum rate, mm/min)
$112 = 2500.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$120 = 75.000    (X-axis acceleration, mm/sec^2)
$121 = 150.000    (Y-axis acceleration, mm/sec^2)
$122 = 150.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$124 = 50.000   
$130 = 1200.000    (X-axis maximum travel, millimeters)
$131 = 1500.000    (Y-axis maximum travel, millimeters)
$132 = 125.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 360.000   
$140 = 0.010   
$141 = 0.010   
$142 = 0.010   
$143 = 0.000   
$144 = 0.000