Schildkroet/GRBL-Advanced #29: linear 4th axis

isensewa, 2020-03-04:
Hi Schilddkroet, I'am just running a cnc-mill (modified shapeoke) with an arduino-uno and grbl. I build also a cnc-foamcutter. I'am planning to go 32bit for both and found your thread. For that I have some questions. For mill: is it possible to have an output which signalize lasermode? reason: I would like to switch between 2 differnt powersources and pwm-voltage for laser and spindle by switching the mode. By the way, have I still to boot when switching the mode laser/ spindle? Another thing: i have 2 x-motors. For homing-purpose I would like to have a home-switch for each motor(automatic squaring the gantry) after homing thy should get the same informations.
For my foamcutter i need a 4th linear-axis. I found in one of the config-files to define the number of linear-axes. But I am not shure if that will work. If it would be possible, i would need a separat homing-switch input for that. I hope you can clarify my questions. Thanks

isensewa, 2020-03-06:
Hi Schildkroet, after reading the STm32nucleo-user-manuals I can answer the questions by myself.
In main you ported the GRBL-Code to a 32bit processor, which has female connetors with the same 
pin-layout as the arduino UNO, so one can use the same shields he had used on the UNO. So my 
questions would be features that are not build in by now. If you see a chance or are willing
to implement these, it would be very nice. But anyway, thanks for porting GRBL to 32bit.

closed 2020-03-06 by isensewa

Schildkroet, 2020-03-07:
Hello,
i'm not sure if i understood your question completely, but this version is basically a port to STM32 with some additional features like backlash compensation, canned cycles and rotary axis.
That's all. There are no additional linear axis, modes, outputs etc.