Schildkroet/GRBL-Advanced #30: Backlash Enabled on X and Y axis causes G2 circle to pause every 90 deg

dharders, 2020-03-10:
Hi, thanks for the great backlash feature!

Just curious though, when I **enable backlash** and run a simple gcode such as:

```
G17
G21
G90
G0 X0 Y20
G2 X0 Y20 I20 K20 F2000
```

The 40mm diameter circle that is meant to be drawn in a continuous smooth motion stops every 90deg. This only happens with **backlash enabled**. Without backlash enabled it works smoothly as expected.

The 90deg stops make sense as this is where each axis changes direction and injects the backlash steps, but isn't the planner meant to optimize the trajectory so it doesn't stop?

If I only enable backlash on the X axis, then there is only 1 stop at the 180deg point when the X axis changes direction.

Have I done something wrong, or are you aware of this and it is a limitation of this implementation?

Thanks heaps!
Dale.

Schildkroet, 2020-03-10:
Hello,
yes, i'm aware of this "problem". This is caused by backlash compensation.
If an axis changes it's direction, a short G0 move is internally inserted to perform backlash compensation. But only the compensated axis is moved (others are 0). That's why there is a short stop in the motion.
So far i didn't came up with a better solution, but i have some ideas to test to improve this behaviour (if i find some time)

closed 2020-03-10 by Schildkroet

dharders, 2020-03-10:
@Schildkroet thanks for reaching out!

I'm working on a solution now too. It's not so easy for steep junctions / arcs at the firmware level. Please keep me updated if you figure something out and I'll do the same! Cheers

hitchhiker85, 2020-03-24:
i have the same *problem*.. any news?

thanks!