Schildkroet/GRBL-Advanced #31: Possible to get it work on STM32F104?

hitchhiker85, 2020-03-16:
Hi ich weis das gehört hier nicht hin... aber ich wollte dich mal fragen ob es eine Möglichkeit gibt eine Version zu bekommen die  für einen STM32F104 wäre?

Wenn ich die Hex flash funktioniert das natürlich nicht mit der USB anbindung...

Danke! und Sorry das ich hier nerve ;)

Schildkroet, 2020-03-16:
Diese Version ist für einen F4, welcher einen anderen Core wie die F1 hat. Es gibt aber diverse Portierungen für den F103, vlt funktionieren diese Versionen sofort oder mit minimaler Anpassung.

closed 2020-03-16 by Schildkroet

hitchhiker85, 2020-03-16:
ok ! danke! wo find ich solche portierungen???? danke!!

Schildkroet, 2020-03-16:
Such einfach nach "grbl stm32" oder "grbl stm32f103". Dort findet sich ziemlich viel.

hitchhiker85, 2020-03-16:
ja ohne 4. achse oder ohne backlash :( 

leider dauert es ewig ein f4 zu bekommen.. tja muss ich eben warten ;) danke dir trotzdem für den support!

Schildkroet, 2020-03-16:
Kannst es ja auch selbst für den F104 porten. Musst im Prinzip nur den HAL Ordner anpassen und die SPL austauschen. Sollte in 1 Tag machbar sein. Alternativ einen F411 bei deinem Distributor des Vertrauens bestellen. Sollte bis Freitag da sein.

hitchhiker85, 2020-03-16:
1. keine ahnung ;) komm noch aus der normalen arduino generation...
2. reichelt sollte es bis mittwoch schaffen ;)
3. danke nochmal! sehr cool von dir für deine hilfe! bester mann!