Schildkroet/GRBL-Advanced #32: A Achse? error 20..

hitchhiker85, 2020-03-20:
Sorry... ich  hab nun mein f411 ;)

aber irgendwie kann ich die A Achse nicht benutzen.. was mach ich falsch?

g91
g1 a200 f100

und dann kommt error 20...

gibts eine alternative kontaktmöglichkeit_?

Schildkroet, 2020-03-20:
Hast du `#define USE_MULTI_AXIS` in Config.h aktiviert?

hitchhiker85, 2020-03-20:
ja!

und auch neu geflashed!
gibts da ein bug?!

hitchhiker85, 2020-03-20:
Moment!

Ich hab mal alles ausgeschaltet und vom netzgenommen, usb abgeklemmt, anderen usb port genommen, wieder Spannung drauf, jetzt gehts, musste aber mit $X freischalten.

normal?

danke für deine hilfe!

Schildkroet, 2020-03-20:
Ja, Homing ist standardmäßig ein.

closed 2020-03-20 by Schildkroet

hitchhiker85, 2020-03-20:
ah super danke!

Schildkroet, 2020-03-20:
Momentan sind keine großen Änderungen geplant. Backlash für A/B macht keinen Sinn, Wie soll eine lockere Achse bearbeitet werden??

hitchhiker85, 2020-03-20:
Wenn man z.b. Umkehrspiel hat durch Schneckenwelle und Rad? weist was ich meine?

hitchhiker85, 2020-03-22:
nabend!!!

sagmal gibts irgendwo eine einstellung um das $x nicht immer eingeben zu müssen?? danke

Schildkroet, 2020-03-22:
Ja, hab doch geschrieben das Homing aktiviert ist

hitchhiker85, 2020-03-22:
ja aber kommentiere ich das ändert sich leider nichts daran. ich muss trotzdem immer freischalten. versteh ich nicht sorry

Schildkroet, 2020-03-22:
Du musst die Settings im EEPROM auch aktualisieren
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#rst-rst-and-rst--restore-grbl-settings-and-data-to-defaults

hitchhiker85, 2020-03-22:
ja! hab ich!

könntest du mir bitte sagen welche werte in der config geändert werden müssen??

$22=0 hab ich auch!

danke

Schildkroet, 2020-03-22:
Wenn $22=0 ist, dann kommt die Meldung nicht mehr. Gerade selbst am Board getestet

hitchhiker85, 2020-03-22:
i h hab in der defaults.h 22=0

in der config.h //#homing_init_lock

was ist dann noch falsch?!

danke

hitchhiker85, 2020-03-22:
hab das // entfernt nun gehts!

danke danke danke:) sry fürs nerven :)

Schildkroet, 2020-03-22:
Was ist die Ausgabe bei `$$` ?

hitchhiker85, 2020-03-22:
22=0

hab den eeprom gewiped per grbl plugin im cambam.

einzige sache die mich wundert ist:
in cambam zieht er sich die $$ nicht alleine.

jeder andere arduino mit grbl macht das.

weis ja nicvt ob du cambam mit dem plugin kennst.

welches cam programm nutzt du?

danke