Schildkroet/GRBL-Advanced #33: port to STM32F411CEU6

creisey, 2020-03-28:
how hard is it to port your code to the black pill. its basicly the same core STM32F411....?

STM32F411CEU6

Schildkroet, 2020-03-28:
Theoretically it should run out the box. But you might run out of pins if you use extended functionality.

creisey, 2020-03-31:
when i flash the .bin file to it, it cant boot the code. only a factory reset will relieve it again.

Schildkroet, 2020-03-31:
OK, might be because of the smaller size of the flash. Settings are stored in last sector of flash, which is missing here.
Try commenting out the Settings_Init-Function and try again.

klembaris, 2020-04-02:
STM32F411CEU6 has 512 KB Flash and WeAct bootloader, but after .hex load there is no connection and com port.

Schildkroet, 2020-04-02:
how do you flash the hex?

klembaris, 2020-04-02:
With STM32 st-link
[image]

Schildkroet, 2020-04-02:
Go to HAL/STM32/stm32f4xx.h and change all HSE_VALUE from 8 to 25MHz and recompile

klembaris, 2020-04-02:
Its same, not working.

Schildkroet, 2020-04-02:
Are you sure you use UART2 on PA2/3?

klembaris, 2020-04-02:
Now yes:), Usb to TTL dongle to ports A2 and A3. But still no response from controller.

Schildkroet, 2020-04-02:
Connect st-link, Select Debug Target in Embitz and start a debug session to see if it runs

klembaris, 2020-04-02:
Debugger-target tab: empty no logs
Debugger tab:

Building to ensure sources are up-to-date
Build succeeded
Selecting target: 
Debug
Adding file: bin\Debug\GRBL_Advanced.elf
Adding source dir: C:\Users\John\Desktop\GRBL-Advanced-software\
Adding source dir: C:\Users\John\Desktop\GRBL-Advanced-software\
Starting debugger: 
done
Setting breakpoints
EmBitz embedded firmware debugger for arm-none-eabi (Sep 30 2016/14:16:49).
Based on (GDB)  7.8.2 - License GPLv3+ by Gerard Zagema.
This program comes with ABSOLUTELY NO WARRANTY; for details type "show war".
This is free software, and you are welcome to redistribute it
under certain conditions; type "show cop" for details.

Reading symbols from bin/Debug/GRBL_Advanced.elf...
done.
Source directories searched: C:/Users/John/Desktop/GRBL-Advanced-software;$cdir;$cwd
Remote debugging using localhost:4242
0x000000000x00000000 in ?? ()
Connected
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...

Schildkroet, 2020-04-03:
Can you debug this programm?
[TestF411CE.zip](https://github.com/Schildkroet/GRBL-Advanced/files/4425490/TestF411CE.zip)

klembaris, 2020-04-03:
Sure, now looks like this:
[image]

Schildkroet, 2020-04-03:
OK, another try with grbl-advanced:
Make sure HSE_VALUE is set to 25000000 in HAL/STM32/stm32f4xx.h

In HAL/STM32/system_stm32f4xx.c set PLL Parameters:

`/************************* PLL Parameters *************************************/
/* PLL_VCO = (HSE_VALUE or HSI_VALUE / PLL_M) * PLL_N */
#define PLL_M      25
#define PLL_N      168

/* SYSCLK = PLL_VCO / PLL_P */
#define PLL_P      2

/* USB OTG FS, SDIO and RNG Clock =  PLL_VCO / PLLQ */
#define PLL_Q      8`

Debug again.

klembaris, 2020-04-03:
Everything works now. Thanks.

[image]

closed 2020-04-03 by Schildkroet

creisey, 2020-05-15:
sorry for my late reply. had to finalize the exposé for my master... i could also flash the firmware according your explanation. is the grbl only working through ttl dongle? or is there also a virtual com port solution? like on the bluepill?

Schildkroet, 2020-05-15:
Only USART. No VCP.

duembeg, 2020-06-13:
Hi, 
Just wanted to ask since the device (blackpill) is advertised as 100Mhz;
wondering if PLL_N should be 192 and SystemCoreClock = 100000000?

hunabapr, 2021-04-14:
I get the same issue, ie "STM32F411CEU6 has 512 KB Flash and WeAct bootloader, but after .hex load there is no connection and com port"

I've followed this up to 3 Apr 2020 and was getting the same results including the connect to the test prog.
But when I make the changes re:HAL/STM32/stm32f4xx.h plus...   I still can't get this to work

Any ideas please?

Taken from ST-Link utility
14:31:15 : Device flash Size : 512KBytes
14:31:15 : Device family :STM32F411xC/E

The best   Paul

hitchhiker85, 2021-09-13:
> Everything works now. Thanks.
> 
> [image]

hi you use the blackpill with stm32f411ce right?

possible to share your file???? thanks

hitchhiker85, 2021-09-15:
> Only USART. No VCP.

is it realy not possible to use vcp?

i convert all to stm32f411ceu6.
gpios... hse.  it works fine with uart pa2 pa3.
but with the usb plug it would be perfect