Schildkroet/GRBL-Advanced #35: Trying to use your grbl for laser engraving, and have pwm issues.

rotski, 2020-04-19:
I have problem with the laser firing when its not supposed to, any idea what i need to change in the firmware?
Those thin diagonal lines, it seems to be random on each time i run the gcode.

[image]

Schildkroet, 2020-04-19:
Pls post the output of $$ and $I
Can you attach the gcode?
How is your laser connected? Schematic?

rotski, 2020-04-19:
$$
$0=10 Step pulse, microseconds (10)
$1=25 Step idle delay, milliseconds (25)
$2=0 Step port invert, mask (0)
$3=0 Direction port invert, mask (0)
$4=0 Step enable invert, boolean (0)
$5=0 Limit pins invert, boolean (0)
$6=0 Probe pin invert, boolean (0)
$10=3 Status report, mask (1)
$11=0.020 Junction deviation, mm (0.020)
$12=0.020 Arc tolerance, mm (0.020)
$13=0 Report inches, boolean (0)
$14=0  
$20=0 Soft limits, boolean (0)
$21=0 Hard limits, boolean (0)
$22=1 Homing cycle, boolean (0)
$23=1 Homing dir invert, mask (0)
$24=100.000 Homing feed, mm/min (100.000)
$25=500.000 Homing seek, mm/min (1000.000)
$26=250 Homing debounce, milliseconds (250)
$27=1.000 Homing pull-off, mm (1.000)
$30=255 Max spindle speed, RPM (255)
$31=0 Min spindle speed, RPM (0)
$32=1 Laser mode, boolean (1)
$100=100.000 X steps/mm (80.000)
$101=80.000 Y steps/mm (80.000)
$102=400.000 Z steps/mm (80.000)
$103=10.000  
$104=10.000  
$110=10000.000 X Max rate, mm/min (5000.000)
$111=10000.000 Y Max rate, mm/min (5000.000)
$112=1000.000 Z Max rate, mm/min (5000.000)
$113=10000.000  
$114=10000.000  
$120=800.000 X Acceleration, mm/sec^2 (800.000)
$121=800.000 Y Acceleration, mm/sec^2 (800.000)
$122=30.000 Z Acceleration, mm/sec^2 (5000.000)
$123=100.000  
$124=100.000  
$130=400.000 X Max travel, mm (400)
$131=300.000 Y Max travel, mm (300)
$132=500.000 Z Max travel, mm (200)
$133=360.000  
$134=360.000  
$140=0.000  
$141=0.000  
$142=0.010  
$143=0.000  
$144=0.000 

$I
[VER: 1.1 Apr 18 2020:]
[OPT:VNZL,63,200]

Here is the gcode, but when i run it with the arduino nano grbl, there is no issues.
https://drive.google.com/file/d/1-iGWt5l_sq6ZDLIsxFbXQyywb4UI3F5d/view?usp=sharing

The 
[image]

This is the current wiring, but the i got the same error when using a cnc shield, did this to try to eliminate the shield as the problem. Thats why i suspect there is something in the firmware.

Schildkroet, 2020-04-19:
Do these lines appear randomly or always at the same position?

rotski, 2020-04-20:
Yes its random.
 Same gcode created from t2 laser and lasergrbl.
[image]
[image]

Schildkroet, 2020-04-20:
The gcode looks good. It seems the laser is not turned off properly at some points.
You connected PWM out (PA_7) to the mana board right? And the laser is controlled only by PWM?

rotski, 2020-04-20:
PA7 yes to the pin corresponding to the Arduino D11 pin on the mana board. 
Its only controlled by pwm. Is there some PWM frequancy that is different to your grbl in comparison to "regular" grbl?

Schildkroet, 2020-04-20:
Well, it should be the same.
Go to SpindleControl.c, Line 104: TIM_Cmd(TIM1, DISABLE);
Copy and paste this command a few times, so it is called serveral times. See if it helps.

rotski, 2020-04-20:
I tried adding it 5 and 20 times,  no notable difference. I will return to use the nano, as i dont really have the time to troubleshoot. Thanks for your help! Will probably try your grbl for a mill some time in the future, i am sure it will work flawlessly. Again, thanks for taking the time to help.

Schildkroet, 2020-04-21:
I think i found the problem.
Please add `TIM1->CCR1 = 100;    // Disable PWM. Output voltage is zero.` after Line 104 and try again.

rotski, 2020-04-21:
Oh, sweet, I will try as soon as I can. Can you post the lines above and below what you want me to add, how they are supposed to look, so I don't mess up.

Schildkroet, 2020-04-22:
It should look like this in Spindle_SetSpeed at Line 105++
[image]

rotski, 2020-04-22:
It works now, thank you very much!

closed 2020-04-22 by Schildkroet