Schildkroet/GRBL-Advanced #36: list board

qiwalker, 2020-04-22:
hi... can i use others STM32 board such as
STM32F103 [ARM Cortex M3] or The STM32F407 [ARM Cortex M4] ?

thanks ^^

qiwalker, 2020-04-22:
hi...one more question, where i can find pin limit for A and B axis ?

Schildkroet, 2020-04-22:
Yes, it is possible, but you have to port some code.
https://github.com/Schildkroet/GRBL-Advanced/issues/33
https://github.com/Schildkroet/GRBL-Advanced/issues/12
https://github.com/Schildkroet/GRBL-Advanced/issues/8

Currently there are no limit pins for A&B. But they can be easily added.

qiwalker, 2020-04-22:
thanks for your explanation,

any tutorial to add limit pin for A & B ?

Schildkroet, 2020-04-22:
No, but should be enough to initialize pins in HAL/GPIO and add handling in grbl/Limits

qiwalker, 2020-04-22:
ok i will try add code in those file, thank you very much...

qiwalker, 2020-04-22:
hi again

at limit.c i add
	limit_state = (GPIO_ReadInputDataBit(GPIO_LIM_X_PORT, GPIO_LIM_X_PIN)<<X_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Y_PORT, GPIO_LIM_Y_PIN)<<Y_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Z_PORT, GPIO_LIM_Z_PIN)<<Z_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Z_PORT, GPIO_LIM_Z_PIN)<<A_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Z_PORT, GPIO_LIM_Z_PIN)<<B_LIMIT_BIT);

and at GPIO i add

#define GPIO_LIM_A_PORT			GPIOA
#define GPIO_LIM_B_PORT			GPIOA
#define GPIO_LIM_A_PIN			GPIO_Pin_0
#define GPIO_LIM_B_PIN			GPIO_Pin_1

is it correct ?

Schildkroet, 2020-04-22:
limit_state |= (GPIO_ReadInputDataBit(**GPIO_LIM_Z_PORT**, **GPIO_LIM_Z_PIN**)<<A_LIMIT_BIT);
limit_state |= (GPIO_ReadInputDataBit(**GPIO_LIM_Z_PORT**, **GPIO_LIM_Z_PIN**)<<B_LIMIT_BIT);

Only if you set the correct defines

qiwalker, 2020-04-22:
Correction
sorry, my mistake

	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Z_PORT, GPIO_LIM_A_PIN)<<A_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_Z_PORT, GPIO_LIM_B_PIN)<<B_LIMIT_BIT);

is there any file need to edit sir ?

qiwalker, 2020-04-22:
Correction
Sorry, my mistake

	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_A_PORT, GPIO_LIM_A_PIN)<<A_LIMIT_BIT);
	limit_state |= (GPIO_ReadInputDataBit(GPIO_LIM_B_PORT, GPIO_LIM_B_PIN)<<B_LIMIT_BIT);

is there any file need to edit sir ?

Schildkroet, 2020-04-22:
> No, but should be enough to initialize pins in **HAL/GPIO** and add handling in **grbl/Limits**

And PA-0 and 1 are already occupied

qiwalker, 2020-04-22:
Done Sir i change to PA-11 and 12

closed 2020-04-29 by Schildkroet

datdadev, 2024-07-18:
Can you share the ported version of STM32F407VET6 chip?