Schildkroet/GRBL-Advanced #38: Baud rate

steph79260, 2020-06-14:
Hello,

I don't understand why, but the baud rate must be set to 57600 for this to work. 
Yet we have 115200 in the config.h ...

Do you have any idea where this can come from?

Thank you and have a nice day, Stéphan

     -  My card : "NUCLEO F-411RE"
     -  Compiled with : "EMBITZ 1.11"
     -  Load with : "STM32 ST-LINK Utility"

Schildkroet, 2020-06-14:
Your board runs only with half speed. Did you modify the clock source?

steph79260, 2020-06-14:
No I didn't touch anything at the file source.
Maybe a jumper on the board ? But i've modifie nothing...

Schildkroet, 2020-06-14:
I meant the board itself

steph79260, 2020-06-14:
No

Schildkroet, 2020-06-14:
Do you know how to debug?

steph79260, 2020-06-14:
I can try...

Schildkroet, 2020-06-14:
Set a breakpoint in HAL/STM32/system_stm32f4xx.c Line 398:
 (if (HSEStatus == (uint32_t)0x01))

Then single step to see if it continues in the following lines or if it jumps into else path in line 440

steph79260, 2020-06-14:
Sorry, the breakpoint is well positioned but then I don't know where to go to debug this line ...

Schildkroet, 2020-06-14:
Select "Debug"-Target at top, then press F8. Programm should stop at breakpoint. Press F10 for single step

steph79260, 2020-06-14:
It's going to the line 401...

[image]

Schildkroet, 2020-06-14:
Ok, delete old breakpoint, set new breakpoint in SPL/src/stm32f4xx_rcc.c Line 1333. If programm stops in main, press F5, then it should stop at breakpoint. Hover mouse over HSE_VALUE. What value is displayed?

steph79260, 2020-06-14:
When I do an F8 after putting a breakpoint line 1330 (stm32f4xx_rcc.c), it jumps directly to line 61 of the main.c ...

[image]

Schildkroet, 2020-06-14:
Thats why i said you should press F5

steph79260, 2020-06-14:
[image]

Schildkroet, 2020-06-14:
Go to HAL/STM32/stm32f4xx.h Line 148 and change 16 to 8

steph79260, 2020-06-14:
[image]

I change ((uint32_t)16000000) by ((uint32_t)8000000) ?

Schildkroet, 2020-06-14:
yes

steph79260, 2020-06-14:
Ok and after i compile and i load on the card ?

Schildkroet, 2020-06-14:
yes

steph79260, 2020-06-14:
Ok to 115200 !!
Thank you very much ;)
Steph

closed 2020-06-14 by Schildkroet