Schildkroet/GRBL-Advanced #40: cambam grbl plugin

hitchhiker85, 2020-07-06:
Das grbl plugin in cambam erkennt nicht die grbl version bzw will sich nicht verbinden!
hast du eine lösung?

Schildkroet, 2020-07-06:
Die Ausgabe ist identisch zu grbl. Andere GCode Sender wie UGS funktionieren bei mir auch problemlos. Daher muss es an irgendeiner speziellen Sache von diesem Plug-In liegen. Dementsprechend solltest du beim Ersteller dieses Plug-Ins nachfragen.  Ich kann dazu leider nichts sagen.

closed 2020-07-06 by Schildkroet

hitchhiker85, 2020-07-06:
ich hab einfach  mal geguckt! sobald ich den reset knopf betätige am f411re funktioniert es!

gibt es eine zeile die ich ändern kann ohne den reset drücken zu müssen?
danke

Schildkroet, 2020-07-06:
Nein, sobald das Board mit Spannung versorgt wird, startet es automatisch. Reset ist nur nach einem Flashvorgang nötig. Man könnte noch die Firmware vom Debugger updaten, da einige damit wohl Probleme hatten

hitchhiker85, 2020-07-06:
weist du auf welche version?

ich muss definitiv jedesmal den reset drücken. auch im ugs. Danke

Schildkroet, 2020-07-06:
Auf die neuste Version

hitchhiker85, 2020-07-06:
hab ich gemacht, auch die letzte Version deiner hex Datei mit §rst=$.
es bleibt gleich.

noch eine Idee?
bekomme nun auch error 2.

danke

Schildkroet, 2020-07-06:
Nein, nicht grbl. Du must die Firmware vom Debugger auf dem NucleoBoard updaten.

hitchhiker85, 2020-07-06:
ich habe beides gemacht.

keine ahnung.
win updates sind deaktiviert.
was kann ich noch machen?
debugger ist grupdated und die neuste version grbl von dir.

ich hab nix verändert. mich wundert das. danke

hitchhiker85, 2020-07-14:
ich hab jetzt mehrere programme ausprobiert.
ich muss tatsächlich immer reset drücken damit alles funktioniert.

hast du echt kein tipp?

hitchhiker85, 2020-07-14:
oder weist du wie ich bei verbindung den reset automatisch hab?

irgend ein bit was gesetzt werden muss? nur im stcube find ich nichts.

hitchhiker85, 2020-07-19:
problem gefunden!

es waren 2 0er Widerstände mit kalten lötstellen.
jetzt geht alles!