Schildkroet/GRBL-Advanced #41: spindle control with Relay?

hitchhiker85, 2020-07-19:
hi! can you please say me what i must cange to control my spindle with relay?

i will only switch on and off.

thanks

Schildkroet, 2020-07-20:
Update to newest version.
Use PB7/Spindle_Enable (See pinout_left) to control a relay.

closed 2020-07-20 by Schildkroet

hitchhiker85, 2020-07-20:
i use pb7 but what must change in config?
thanks for your help!

Schildkroet, 2020-07-20:
Nothing, except invert mask if neccessary

hitchhiker85, 2020-07-20:
i dont need changes if i use relay on pb7?

and which values i must set i my cam tool?
m3 m5 nothing hapens.
with fresh hex file.

hope you can help me.

thanks

Schildkroet, 2020-07-20:
No changes. M3 turns it on, M5 off. I tested this right now. Works perfectly

hitchhiker85, 2020-07-20:
ok tomorrow i will erase the chip complete and try it again.

i use a cheap single relay shield.
i think i must change the high/low on/off option..

and spindle speed i dont change ? i mean $30?

thanks!