Schildkroet/GRBL-Advanced #45: NUCLEO STM32F446RE problem

geertmalfait, 2020-09-04:
Hi,
I try to get GRBL-Advanced up and running on a NUCLEO STM32F446RE.
I followed the instructions in the Readme.md to  build the software on a Linux system. 
After making following adaptions to the Makefile:
-replaced the compile options -DSTM32F411xE -DSTM32F411RE by -DSTM32F446xx
-replaced -T../stm32f446re_flash.ld by -T../stm32f411re_flash.ld
I was able to "make all" and "make flash"
After flashing and rebooting the following appears on the serial port when pushing Enter( with minicom /dev/ttyACM0 115200 8N1):
-ok
-error:9
then the communication hangs...
Any suggestions?

Schildkroet, 2020-09-05:
Hello,
pls change 16 to 8 in the following lines:
https://github.com/Schildkroet/GRBL-Advanced/blob/c30489527af4f9c052eadb368d36a46b09ed8bde/HAL/STM32/stm32f4xx.h#L147-L149
https://github.com/Schildkroet/GRBL-Advanced/blob/c30489527af4f9c052eadb368d36a46b09ed8bde/HAL/STM32/system_stm32f4xx.c#L166-L169

geertmalfait, 2020-09-07:
Hi, Your suggestion solved the issue, Thanks!

One more thing: I'm considering porting GRBL-Advanced to STM32cubeIDE. Any suggestions about this? (e.g. is the IO and clock configuration documented or should I go through the sources?)

Schildkroet, 2020-09-07:
No extra documentation. Look into source code

closed 2020-09-07 by Schildkroet