Schildkroet/GRBL-Advanced #47: Abort/Hold pins

LaurentLM, 2020-10-13:
Hi,
Thanks for the good work, I have been using this for a mini mill for a while and it's great.
I am using a Nucleo F401RE and when wiring new buttons recently, I noticed the abort/hold pins were inverted compared with standard Arduino GRBL. 
I have Abort on A1 & Hold on A0. I cannot find where this is defined in the code.

I also noticed they were not pulled up, but that's easy to change (line 187 of gpio.c).
Not a huge deal, but I thought I'd report it.

Schildkroet, 2020-10-13:
Hello,
the control mask for inverting buttons is in
https://github.com/Schildkroet/GRBL-Advanced/blob/48cd2aa65596d6a5345bc55f2344d25d993ab9e7/grbl/util.h#L81

The actual reading happens in
https://github.com/Schildkroet/GRBL-Advanced/blob/f1df62453c11b7769aae0a6ac1416a0eba1c3b0c/grbl/System.c#L61-L89

Yes, it is a bit messy right now. Maybe i find some time to fix it

Schildkroet, 2020-10-13:
I fixed the missing invert command. You can now invert your pins as you like.
These are the corresponding bits
https://github.com/Schildkroet/GRBL-Advanced/blob/69a3cb81a4d2d1876fe689bc3f9c87dfba74984b/grbl/Settings.h#L34-L40

You can set with with DEFAULT_SYSTEM_INVERT_MASK and $0=xxx

closed 2020-10-13 by Schildkroet

LaurentLM, 2020-10-14:
Thanks for looking into this. 
I flashed the latest commit and it fixed most of my issues. 

Here is what's happening:
I changed this line: 
https://github.com/Schildkroet/GRBL-Advanced/blob/69a3cb81a4d2d1876fe689bc3f9c87dfba74984b/HAL/GPIO/GPIO.c#L187
to `GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;` to match standard GRBL's pull-ups on A0, A1, A2 pins.

The Feed Hold pin (A1), behaves differently from the other two and is active when nothing is connected to it (it reads 3.3V on the voltmeter). Cycle start and Abort pins also read 3.3V but are not registered as active unless pinned down to ground (expected behaviour).

I suspect this is what the DEFAULT_SYSTEM_INVERT_MASK is for. Could you explain how it works? I tried different values and cannot get it to work.

LaurentLM, 2020-10-14:
Thanks for looking into it, $0=0 sorted the problem.
Just out of curiosity, what happened to the "Step pulse, microseconds" setting that $0 used to control? 

Also, would it make sense to change this:
https://github.com/Schildkroet/GRBL-Advanced/blob/69a3cb81a4d2d1876fe689bc3f9c87dfba74984b/HAL/GPIO/GPIO.c#L187
into `GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;` ?
That would match the standard GRBL behaviour.

Schildkroet, 2020-10-14:
I set it to a fixed value. All common drivers work the same, so i don't think it is necessary.

LaurentLM, 2020-10-15:
Makes sense. Thanks again or the great work!