Schildkroet/GRBL-Advanced #49: STM32F446RE stalling

McCodie, 2020-11-27:
Hi, did you really test it with the STM32F446RE? Because the communication just seems to stall after a few commands.
I tried to build the firmware with MacOS & Windows, both gave me the same result.

$H let's it crash immediately (nothing is connected to the Nucleo board).

Schildkroet, 2020-11-27:
Yes, I'm actually running it on a F446.
What do you mean by crash? Whats the raw terminal ouput? How did you compile it?

McCodie, 2020-11-27:
Hi,

thanks for your quick reply!

I somewhat think the problem is the $J parameter which is submitted by the universal gcode sender.
Which application do you use for submitting the gcode? Candle?

>>> $G
[GC:G0 G54 G17 G21 G90 G94 G98 M5 M9 T0 F0 S0]
ok
>>> $X
[MSG:Caution: Unlocked]
ok
>>> $$
$0 = 0    (Step pulse time, microseconds)
$1 = 50    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.001    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 3000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$103 = 10.000   
$104 = 10.000   
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 10000.000   
$114 = 10000.000   
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 30.000    (Z-axis acceleration, mm/sec^2)
$123 = 100.000   
$124 = 100.000   
$130 = 10.000    (X-axis maximum travel, millimeters)
$131 = 50.000    (Y-axis maximum travel, millimeters)
$132 = 500.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 360.000   
$140 = 0.010   
$141 = 0.010   
$142 = 0.010   
$143 = 0.000   
$144 = 0.000   
ok
>>> $J=G21G91Z-500F1
ok
>>> $J=G21G91Z-500F1
ok
>>> $J=G21G91Z-500F1
ok
>>> $J=G21G91Z-500F1
ok
>>> $J=G21G91Z-500F1
ok
>>> $J=G21G91X500F1
ok
>>> $J=G21G91X500F1
>>> $J=G21G91X500F1
>>> $J=G21G91X500F1
>>> $J=G21G91X-500F1
>>> $J=G21G91X-500F1

Schildkroet, 2020-11-27:
How do you compile it?

Try this HEX. Restore settings after flashing
[GRBL_Advanced.zip](https://github.com/Schildkroet/GRBL-Advanced/files/5608537/GRBL_Advanced.zip)

McCodie, 2020-11-27:
I compiled it with Embitz 1.11 (Windows), also on MacOS using the Makefile, both gave me the same result.

I modified the Makefile a little bit.
CFLAGS      :=  -O2 -g1 -std=c11 -Wall -Wextra $(INCLUDE) -fno-common -fsingle-precision-constant -fdata-sections -ffunction-sections -fomit-frame-pointer -mlittle-endian  -DUSE_STDPERIPH_DRIVER -DSTM32F446RE -DSTM32F446xx -D__FPU_USED -DARM_MATH_CM4 -Wimplicit-fallthrough=0
#CFLAGS      :=         -O2 -g1 -std=c11 -Wall -Wextra $(INCLUDE) -fno-common -fsingle-precision-constant -fdata-sections -ffunction-sections -fomit-frame-pointer -mlittle-endian  -DUSE_STDPERIPH_DRIVER -DSTM32F411xE -DSTM32F411RE -D__FPU_USED -DARM_MATH_CM4 -Wimplicit-fallthrough=0

Schildkroet, 2020-11-27:
Did you also change the linker flags?
https://github.com/Schildkroet/GRBL-Advanced/blob/software/Makefile#L40

McCodie, 2020-11-27:
I tested your hex file and it's not working.
yes I also modified the LDFLAGS (as mentioned I also switched to STM32F446 in Windows and the result was exactly the same as with the macos binary).

LDFLAGS         :=      -lm -flto -Wl,--gc-sections -T../stm32f446re_flash.ld -Wl,-Map=$(OUTPUT).map --specs=nosys.specs -nostartfiles --specs=nano.specs

Is this a stock NUCLEO-F446RE board? Did you install a crystal on X3?

Schildkroet, 2020-11-27:
> I tested your hex file and it's not working.
Not working at all? Garbage output? Same behaviour as yours?

McCodie, 2020-11-27:
it's not returning anything via the serial interface did you set the serial to 115kbaud?

st-flash --format ihex write GRBL_Advanced.hex 
the other compiled image I'm loading with:
st-flash write GRBL_Advanced.bin 0x8000000

st-flash 1.6.1
2020-11-28T00:13:38 INFO common.c: F446: 128 KiB SRAM, 512 KiB flash in at least 128 KiB pages.
2020-11-28T00:13:38 INFO common.c: Attempting to write 57748 (0xe194) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 2020-11-28T00:13:38 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Sector:0x1 Size:0x4000 2020-11-28T00:13:39 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Sector:0x2 Size:0x4000 2020-11-28T00:13:39 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Sector:0x3 Size:0x4000 2020-11-28T00:13:39 INFO common.c: Flash page at addr: 0x0800c000 erased
2020-11-28T00:13:39 INFO common.c: Finished erasing 4 pages of 16384 (0x4000) bytes
2020-11-28T00:13:39 INFO common.c: Starting Flash write for F2/F4/L4
2020-11-28T00:13:39 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 24980
2020-11-28T00:13:40 INFO common.c: Starting verification of write complete
2020-11-28T00:13:41 INFO common.c: Flash written and verified! jolly good!

**** Connected to cu.usbmodem14103 @ 115000 baud ****

McCodie, 2020-11-27:
I have 2 NUCLEO-F446RE boards here both show the same behaviour.

Schildkroet, 2020-11-27:
This is the code i'm currently running without problems. I just changed HSE value, since im using 16MHz
[GRBL_Advanced.zip](https://github.com/Schildkroet/GRBL-Advanced/files/5608972/GRBL_Advanced.zip)

Did like 100x jogs and then homed. Since nothing is connected, is fails after about 1min

McCodie, 2020-11-27:
can you show me your settings? $$

Schildkroet, 2020-11-27:
`#define DEFAULT_X_STEPS_PER_MM 			400
  #define DEFAULT_Y_STEPS_PER_MM 			400
  #define DEFAULT_Z_STEPS_PER_MM 			400
  #define DEFAULT_A_STEPS_PER_DEG           10.0
  #define DEFAULT_B_STEPS_PER_DEG           10.0
  #define DEFAULT_X_MAX_RATE 				2400.0  // mm/min
  #define DEFAULT_Y_MAX_RATE 				2400.0  // mm/min
  #define DEFAULT_Z_MAX_RATE 				2400.0  // mm/min
  #define DEFAULT_A_MAX_RATE                12000.0  // °/min
  #define DEFAULT_B_MAX_RATE                12000.0  // °/min
  #define DEFAULT_X_ACCELERATION 			(100.0*60*60) // 30*60*60 mm/min^2 = 30 mm/sec^2
  #define DEFAULT_Y_ACCELERATION 			(100.0*60*60) // 30*60*60 mm/min^2 = 30 mm/sec^2
  #define DEFAULT_Z_ACCELERATION 			(100.0*60*60) // 30*60*60 mm/min^2 = 30 mm/sec^2
  #define DEFAULT_A_ACCELERATION            (200.0*60*60) // 100*60*60 mm/min^2 = 100 mm/sec^2
  #define DEFAULT_B_ACCELERATION            (200.0*60*60) // 100*60*60 mm/min^2 = 100 mm/sec^2
  #define DEFAULT_X_MAX_TRAVEL 				300.0   // mm NOTE: Must be a positive value.
  #define DEFAULT_Y_MAX_TRAVEL 				400.0   // mm NOTE: Must be a positive value.
  #define DEFAULT_Z_MAX_TRAVEL 				900.0   // mm NOTE: Must be a positive value.
  #define DEFAULT_A_MAX_TRAVEL              360.0   // °
  #define DEFAULT_B_MAX_TRAVEL              360.0   // °
  #define DEFAULT_SPINDLE_RPM_MAX 			1250.0  // rpm
  #define DEFAULT_SPINDLE_RPM_MIN 			0.0     // rpm

  #define DEFAULT_X_BACKLASH                0.05    // mm
  #define DEFAULT_Y_BACKLASH                0.0     // mm
  #define DEFAULT_Z_BACKLASH                0.06   // mm

  #define DEFAULT_SYSTEM_INVERT_MASK		0x3F    //
  #define DEFAULT_STEPPING_INVERT_MASK 		0       //
  #define DEFAULT_DIRECTION_INVERT_MASK 	0       //
  #define DEFAULT_STEPPER_IDLE_LOCK_TIME	255     // msec (0-254, 255 keeps steppers enabled)
  #define DEFAULT_STATUS_REPORT_MASK 		1       // MPos enabled
  #define DEFAULT_JUNCTION_DEVIATION 		0.02    // mm
  #define DEFAULT_ARC_TOLERANCE 			0.001   // mm
  #define DEFAULT_REPORT_INCHES 			0       // false
  #define DEFAULT_INVERT_ST_ENABLE 			0       // false
  #define DEFAULT_INVERT_LIMIT_PINS 		0       // false
  #define DEFAULT_SOFT_LIMIT_ENABLE 		0       // false
  #define DEFAULT_HARD_LIMIT_ENABLE 		1       // true
  #define DEFAULT_INVERT_PROBE_PIN 			0       // true
  #define DEFAULT_LASER_MODE 				0       // false
  #define DEFAULT_HOMING_ENABLE 			1       // true
  #define DEFAULT_HOMING_DIR_MASK 			0       // move positive dir
  #define DEFAULT_HOMING_FEED_RATE 			35.0    // mm/min
  #define DEFAULT_HOMING_SEEK_RATE 			600.0   // mm/min
  #define DEFAULT_HOMING_DEBOUNCE_DELAY 	250     // msec (0-65k)
  #define DEFAULT_HOMING_PULLOFF 			2.0     // mm
  #define DEFAULT_TOOL_CHANGE_MODE          1       // 0 = Ignore M6; 1 = Manual tool change; 2 = Manual tool change + TLS`

McCodie, 2020-11-27:
Thanks, I think it is due to some very wrong settings.
Seems like I have some progress, I just ran the setup wizard in Universal GCode Platform and set the default values.

I did not do any real configuration yet, I will wire it up with my machine and do all the real setup.
I'll update this discussion once I'm done with that.

thanks a lot for your quick reply!

Schildkroet, 2020-11-27:
Did you test my last file?

McCodie, 2020-11-27:
yes, same result - so it's very likely due to the wrong settings... as mentioned I have updated it again with my build but ran the  wizard for the settings. It's not stalling anymore now (but I only did some smaller tests)
I will continue tomorrow with the real wiring.

Schildkroet, 2020-11-27:
BTW don't write to $0. It contains the system flag, not step pulse time

McCodie, 2020-11-28:
mm I just did some more tests with it and it's not reliable.

>>> $$
$0 = 14    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 7    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 3650.000    (Z-axis travel resolution, step/mm)
$103 = 10.000   
$104 = 10.000   
$110 = 3000.000    (X-axis maximum rate, mm/min)
$111 = 1500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$113 = 10000.000   
$114 = 10000.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 1.000    (Y-axis acceleration, mm/sec^2)
$122 = 1.000    (Z-axis acceleration, mm/sec^2)
$123 = 100.000   
$124 = 100.000   
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 360.000   
$140 = 0.010   
$141 = 0.010   
$142 = 0.010   
$143 = 0.000   
$144 = 0.000   
ok
>>> $3=0
ok
>>> $J=G21G91Z-1F1
ok
>>> $J=G21G91Z-1F1
ok

**** Canceling file transfer. ****

>>> $J=G21G91Z1F1
ok
>>> $J=G21G91Z1F1
ok

**** Canceling file transfer. ****

>>> $J=G21G91Z1F100
ok
>>> $J=G21G91Z-1F100
>>> $J=G21G91Z-1F100
>>> $J=G21G91Z-1F100
>>> $J=G21G91Z-1F100
>>> $J=G21G91Z-1F100
**** Connection closed ****
**** Connected to cu.usbmodem14203 @ 115000 baud ****
**** Connection closed ****
**** Connected to cu.usbmodem14203 @ 115000 baud ****
Grbl 1.1f ['$' for help]
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$
Grbl 1.1f ['$' for help]
>>> $G
>>> $$

I was only moving up and down the Z axis, I have done some STM32 firmware myself and it's stable on the STM32 controller so either some settings are messing around with it or this just doesn't work reliable at the moment.
I'll go and try linuxcnc next..

Schildkroet, 2020-11-28:
What happens if you use a terminal (like hterm) and enter the commands manually?

closed 2020-12-09 by Schildkroet

McCodie, 2020-12-14:
Hi,

I used coolterm on macos and grbl is crashing after a few commands, I cannot imagine that it will be stable for you if you use the same controller and eg. g-code sender because there aren't that many parameters that can be different.
Did you really cut something with that grbl port?

I have been using linuxcnc for a few days now and there are no such issues.

McCodie, 2020-12-14:
just use universal g-code sender and jog around a few times and it will crash. it makes no difference if you copy / paste the commands a few times.

Schildkroet, 2020-12-14:
I send like 100 cmds with ugs and everything worked fine (Win and Linux). Did you tried with newest commit without changing any settings?

jielemon, 2022-07-16:
> just use universal g-code sender and jog around a few times and it will crash. it makes no difference if you copy / paste the commands a few times.

Hi, which universal g-code sender did you use? Could you please provide the download link?

sundtek, 2022-07-17:
this is old ... the problem were EMC issues.