Schildkroet/GRBL-Advanced #50: G code

Emre-ee, 2020-11-30:
Hi,
My name is Emre. I am working on cnc machine project. I have a question. I am using the grbl library. In addition, I have added it in the encoder and I can compare the position information from the encoder with the position information from the stepper motor. However, if the motor has not reached the desired position, I want to move it in code myself. I know how to move from serial terminal. How do I write the move code? Since I am using Arduino, I could not fully understand the operation of the code because I could not debug the code. Can you please help?

Sincerely,
Mehmet Emre Öztürk

Schildkroet, 2020-12-01:
Hello,
after a move finished (mc_line or mc_arc), you need to check encoder offset, calculate remaining distance and perform another move to get to your desired encoder position. But you have to save your current position (in grbl) and reset it to this value after the move, so position is not changed by this correction move.

Emre-ee, 2020-12-02:
Hİ,
Thank you for answering. I understood what you say. But i didn't understand one thing. Can you give some information about the functions that move the stepper motor? 

Thank you.

Schildkroet, 2020-12-02:
The stepper_isr in stepper.c generates the pulses to move the stepper. To move to a specified position, the easiest way is to call mc_line() with the new target, everything else is done by grbl. But it also updates the system position according to the move, which is not desired for correction moves.
The easiest way would be to use stepper drivers with encoder input, so grbl doesn't have to handle this.

But here is not the right place to talk about such things. You should ask in a dedicated forum

closed 2020-12-02 by Schildkroet

Emre-ee, 2020-12-02:
Ok, thank you very much.