Schildkroet/GRBL-Advanced #51: 500 Khz

IhsanFuadi3, 2020-12-12:
Hi
in the nucleo f446 the speed of each axis can be up to 500 khz?
thanks before

Schildkroet, 2020-12-12:
Hello,
i never tried, but i think it should be possible.
But you have to change max_step_rate https://github.com/Schildkroet/GRBL-Advanced/blob/software/grbl/Config.h#L313-L317

IhsanFuadi3, 2020-12-17:
thank for you reply sir,
I change max_step_rate and F_CPU in util.h,and it's working

Schildkroet, 2020-12-17:
You shouldn't change F_CPU unless you actually change the clock tree. Otherwise grbl won't calculate step rates etc correct

IhsanFuadi3, 2020-12-17:
you mean possible miss step?

Schildkroet, 2020-12-17:
No, wrong feedrate etc

IhsanFuadi3, 2020-12-17:
oo i see,but if i just change max _step_rate,grbl only up to 250 khz,how i increase more than 250 khz?

Schildkroet, 2020-12-17:
Double timer frequency here
https://github.com/Schildkroet/GRBL-Advanced/blob/software/HAL/TIM/TIM.c#L147-L155
https://github.com/Schildkroet/GRBL-Advanced/blob/software/grbl/util.h#L46

IhsanFuadi3, 2020-12-17:
if may i know,how calculate timer frequancy,if I want 500 khz sir?

Schildkroet, 2020-12-17:
what processor do you use?

Schildkroet, 2020-12-17:
`TIM_TimeBaseStructure.TIM_Prescaler = 7-1;		// 24 MHz`

Change to :
`TIM_TimeBaseStructure.TIM_Prescaler = 3;		// 48 MHz`

`F_TIMER_STEPPER             48000000UL`

IhsanFuadi3, 2020-12-17:
i use Nucleo F446 sir

IhsanFuadi3, 2020-12-17:
is there any problem if i set max_step_rate to 500000 Khz sir?with your suggest frequency

Schildkroet, 2020-12-17:
I didn't tried it yet. You have to find out yourself

IhsanFuadi3, 2020-12-17:
Ok Sir,thanks for the reply

closed 2020-12-25 by IhsanFuadi3