Schildkroet/GRBL-Advanced #53: Zwo end stop limits (min, max) for all three axes

engineerlab, 2021-01-12:
Hello Schildkroet,

I'm really happy that something exists in "competition" to smoothieware. I'm really disappointed about smoothieware which promote that it can use for CNC milling machine. Yeah it could use it but for me its useless because lack of of safety and "standard" features for controlling a safe and useful cnc machine...

I hope I have more luck with your firmware, especially the backslash function that can make it charming.

What I'm missing is the pssobility to have two limits (min, max) for the axes, for homing and safety. If I'm right, this is not considered at the moment or I'm wrong? Do you think this possibility could be added?

Greetings engineerlab

Schildkroet, 2021-01-12:
Hello,
i'm already using two limit inputs, but i didn't merged it yet. Maybe i consider merging it for the next release.

phomann, 2021-01-12:
Hi,
Can't you just wire the two limit switches for an axis in series and feed then into the one controller input? If the limit switch is hit nobody cares which particular limit switch is hit, just that the machine stops.
You can also use the same two limit switches for homing. Since the machine knows which direction it is moving an axis during homing, it knows which switch is hit.

In fact, if you home the axes one at a time, you can put all the limit and home switches in series and feed them into a single input on the controller (assuming that the controller allows you to assign multiple functions to a single input. 

Cheers,

Peter

engineerlab, 2021-01-12:
Hi,

@Schildkroet that would be nice :)
@ phomann
Hm you mean wire the two endstop for one axes in parallel? Series would not work, because both must be triggered at the same time. In parallel would work for safety purpose mamybe even for homing if the controller knows the direction for homing, thats correct. Anyway two input is a cleaner solution I think

phomann, 2021-01-13:
Yes if you are using N/O switches then they need to be wired in parallel. It is usual to have all safety/home switches wired as N/C (normally closed). That way, if a wire breaks than the machine will stop rather than crashing into an end stop. That's why I suggested N/C switches.
Using N/O switches for limits may be Ok on small 3D printers or CNC machines, but on a large machine it could be catastrophic.
In my mind using a single input instead of 6 is a clearer solution, but that might just  be how I think. :)

engineerlab, 2021-01-13:
Hi you're right, I looked up my endstop configuration and I'm using normaly closed proximity sensors. In cnc configurations with two motor per axes this would help also I think. Let us decide Schildkroet :) As he said he is using two inputs so this will cause no additional work

commit 880b4d86 (2021-02-07): Code refactoring
Updated makefile with new gcc9
Updated readme
Added Lathe support
Added separated limit inputs (fix #53)
Added encoder support
Added support for dynamic tool length offset on all axis
fix #54
closed 2021-02-07 by Schildkroet