Schildkroet/GRBL-Advanced #55: please contact me. I wish to contribute to your project using a STM32F746 or H750 board

aroyer-qc, 2021-02-10:
aroyer.qc@gmail.com

Instead of working alone to add a graphical interface (mostly done). I want to know if you might be interested to cooperate..
the way i see that.. it won't interfere with the GRBL code, only run into another task ( (yes using a OS) and hook on variable for display.

also it can be possible to run GCode in standalone mode, using virtual uart (I have done this driver). i'm using my own hal, but both system can be run at the same time... i'm also coding using EmBitz!! 

I like your recent code refactoring !

hankecnc, 2021-02-19:
The same price as H750VB, up to 480Mhz

aroyer-qc, 2021-02-19:
I think the power of the new STM32 CPU and use of a GUI can bring GRBL to next level... Atmega is a thing of the past :)
this advanced version.. do much more....  I have on hand the STM32H745I-DISCO and STM32F746G-DISCO. I can also get a STM32H750B-DK..

hankecnc, 2021-02-19:
Yes, Atmega should have been buried in the grave long ago。

For amateur cnc enthusiasts, grbl with stm32 powerful cpu is a good choice, so you can save a lot of money.
I have a budget, DIY a 4030 casting is only 900 US dollars
A set of casting frame with a stroke of 4030 USD 500
1.5kw electric spindle and inverter 130 USD
XYZ ball screw and linear rail $150
XYZ 57 stepper motor and driver $100
[image]

Schildkroet, 2021-02-20:
Hello,
could you be more precise about your intentions and how can i be of any help?
IMO GRBL-Advanced is in a very stable state and i dont intend to make any major changes like RTOS, GUI etc. which might cause a lot of problems.

aroyer-qc, 2021-02-22:
I can understand your skepticism to change the software as I face that often in my professional life. I can do it for myself and prove it's working.. I want to be able to eliminate the need of a GUI running on a PC ... I want to put all those functionality on the controller itself. That can enable me to put gcode file on a SD card and easily choose a file to run on the machine by choosing the file on the board screen. Also the GUI serve as DRO (this can be the first part). **So I have another suggestion**... and the board cost around 100$ US.. it's the STM32H745i-DISCO... it has dual core, a Cortex-M4 running at 240 Mhz and it can run GRBL Advanced as it is without RTOS... without interference... and I can run the GUI on the Cortex M7. the only thing I will have to do is read the variables to display value. The memory can be shared completely and/or protected, or permission can be given to one or the other core.  I can also use serial port to change value like a PC can do it or in future use direct communication between the core... I don't want to play in the GRBL core and have no desire to change it... I just want to add GUI functionality to it.

Schildkroet, 2021-02-25:
The H745i sound like the best solution for you. You can run GRBL on the M4 and the GUI on the M7. The M7 can then talk to GRBL via serial like an "external" pc. No shared memory etc needed.

closed 2021-08-07 by Schildkroet