Schildkroet/GRBL-Advanced #56: Dual motor auto-squaring

Zorki4, 2021-02-18:
Hi Schildkroet,
Can you add support for dual-motor axis?
The only difference is in homing procedure. Slave motor must have additional limit switch, then software can home two motors of one axis independently. I think it should be quite simple to add.
This option is included in grbl 1.1h.
Best Regards.

Schildkroet, 2021-02-20:
Hello,
this feature is currently not planned, but i might add it later if i find some time for it.

Zorki4, 2021-02-24:
Hi. 
I'm not very good at programing and neved used ARM processors, but since grbl-advanced is based on orginal gnea/grbl i'm was able to make it work (just using copy-pase method, and a little bit of tuning to match new processor). Everything seems to work properly. I think that after a few days of testing i'll be able to share modified code.

I'm new to the github, and not shure how it works. Can I add modified version?

Schildkroet, 2021-02-25:
Usually you make a [Pull Request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests)
Or you can send me the source to include the diff

Zorki4, 2021-03-17:
I had to switch back to previous controller, because i had some problems.

There is something wrong with speed of the movement. Machine is decelerating to about zero feed between G-code commands. It is most visible when acceleration is set to low value (100mm/s, I normally run 750mm/s). Circle movement is best example: it is not one full circle, but four quater-circles, so machine slows down every 90 degree.
Same g-code on standard grbl runs OK.

I dont think i have changed anything in the section where speed (feed) is calculated.

Im was not able to create pull request on Schildkroet giihub. I had to make my own fork on my account, and there is pull request. All of the changes i've made are clearly visible there.

Schildkroet, 2021-03-17:
This slow downs are known #30 and are caused by backlash compensation

Zorki4, 2021-03-17:
That was it!
I still need few days to add some #ifdef to GPIO.h and GPIO.c, but i think that the rest is good enough and should work OK.

J-Dunn, 2021-05-23:
Zori, did you ever submit a PR for this? It sounds like it could be useful.

WolfgangBlack, 2021-05-26:
I'm building a cnc router and would love to have squaring for dual Y stepper and a rotary axis....

J-Dunn, 2021-05-27:
> I'm building a cnc router and would love to have squaring for dual Y stepper and a rotary axis....

I looked into this and decided a hardware solution was best. Some stepper motors are available with a spindle protruding both ends. You can then add a 10mm toothed belt and pulleys for minimal cost. You then adjust the relative position of the two ball-screws and lock it up permanently. This means the machine does not go out of true each time you power off or reset. 

Check the lengths of belts available to avoid buying a custom length or adding a tensioning pulley.

J-Dunn, 2021-05-27:
BTW , I think upstream GRBL now has dual motor zeroing , so this could be back ported to this fork.

WolfgangBlack, 2021-12-16:
I am building a QueenBee Pro and would love to autosquare the Y axes and have a rotary axis A.
That would be my ideal setup for this CNC router.
I am using external drivers so would love to use the 4 on the CNC Shield v3.51 for X, Y, Z and pull signals from the Nucleo board to drive a rotary axis later on.
I am using 3d printed covers for the steppers to house 4 pin microphone connectors for the cabling and may have room at the from of the machines for pulleys and a belt to sync them but it is 1500mm apart so not sure how good that would be.

J-Dunn, 2021-12-16:
With a belt that long you will need a means to adjust tension. Probably go for wider than 10mm. My axes are about 70cm apart.

lairdgit, 2021-12-16:
J-Dunn,
I have a question about your machine spindle motor.  Is it speed controlled with PWM?
Thanks, Laird

WolfgangBlack, 2021-12-16:
Autosquaring is used by a lot of people on CNC machines for a few years now.
The MPCNC has dual X and Y motors and relies on it heavily.

J-Dunn, 2021-12-16:
Yes chinese VFD. Good for wood and milling tools but lacks torque for lower speeds like drilling steel or larger daim drills. Drills >6mm stall the spindle. I maybe did not make a well-informed choice there.

lairdgit, 2021-12-16:
> Yes Chinese VFD, 2kW spindle.<
I am more interested to know if you control the speed of the motor with a PWM signal.
Thanks, Laird

J-Dunn, 2021-12-16:
> The MPCNC has dual X and Y motors and relies on it heavily.

Why do they need to "heavily rely" on auto-squaring? Because there is no physical connection. I'm saying a belt is a means to set this up definitively, just like other mechanical alignments. I started working on auto-alignment since I had already added "4th" axis to get independent dual drive ( for other reasons ).

J-Dunn, 2021-12-16:
> I am more interested to know if you control the speed of the motor with a PWM signal.
Sorry, I misunderstood your question. 
Yes, I used PWM through an opto. RC filter on the opto output to VFD. Opto fed with 10V output from VFD and returns 0-5V control signal. ( some voltage drop on opto and resistive divider in filter ).

lairdgit, 2021-12-16:
>  Yes, I used PWM through an opto. ...<
I also use PWM speed control through an opto, but I added issue #75
"CNC spindle motor PWM signal goes off (zero speed) when Backlash compensation makes a correction."
My motor slows down too much to be useful if backlash is used. Did you have a similar experience?
Laird.

J-Dunn, 2021-12-16:
I don't use backlash. I have solid ballscrew drives. 

BTW leave a one line space after a quote to prevent everything getting quoted ;)

lairdgit, 2021-12-16:
I wish I could afford ballscrew drivesI but my machine is a cheap one I use to make prototype PCBs.
I also built this one and it had the same spindle speed problem:
https://github.com/shooter64738/grbl-mega
I spent 3 or 4 days looking at the "Advanced grbl" source code but didn't find out how it was happening.
So I am in the process of adding a secondary CPU, Arduino Nano, to watch for M3, M5, M8 and M9.  I use the coolant signal to stop the spindle when the job is finished.  Otherwise the motor signal gets terminated early because of the gcode buffer etc.  It got turned off when it saw a M5.
Laird

J-Dunn, 2021-12-16:
Where are you going to "watch".  The M* commands come from outside GRBL. It seems your problem is within GRBL if you say it is related to this backlash code.

lairdgit, 2021-12-16:
> Where are you going to "watch". The M* commands come from outside GRBL. It seems your problem is within GRBL if you say it is related to this backlash code.
> 
The STM32F411 Nucleo Board has a connector CN3 with Tx and Rx on it, so I just wire Tx to the Nano.
> Turning off at corners or axis reversal is a laser **option** in GRBL. Have you ensured that is turned off?
<

Yes, $32=0 

This is just a brute force, blast it away with my little Nano I had in the junk box.

I will have to think about how I use laser mode in the future.  Maybe just pass the normal Nucleo PWM signal through the Nano PWM output pin.  Let it follow the Nucleo signal.

J-Dunn, 2021-12-16:
Sure you can monitor the serial comms but if this coming from the backlash code in GRBL-advanced, how does that help? The M codes are not when this is happening. Maybe I just don't follow what you are doing or expecting to do. 

> I will have to think about how I use laser mode in the future.

Why does laser mode come into this? You said you don't want it and it is turned off. 
I use PB0. It pretty simple to set up, then link it to an opto.

Schildkroet, 2021-12-16:
FYI
There was no reply from OP so far. I didn't receive any code changes for this issue.

WolfgangBlack, 2021-12-16:
>There was no reply from OP so far. I didn't receive any code changes for this issue.

Disappointing.....

lairdgit, 2021-12-16:
> Sure you can monitor the serial comms but if this coming from the backlash code in GRBL-advanced, how does that help? The M codes are not when this is happening. Maybe I just don't follow what you are doing or expecting to do.
> 
> > I will have to think about how I use laser mode in the future.
> 
> Why does laser mode come into this? You said you don't want it and it is turned off. I use PB0. It pretty simple to set up, then link it to an opto. There was all sorts of funky swapping stuff around to get these extras stuffed into the very limited IO of Arduino. That makes no sense on STM32.

Hi J-Dunn
I forgot to mention I use the Nano to generate a new stable PWM signal on D9 at 10 KHz. when it gets a M3 Sxxxx.  So I if I use a laser I will have to disable that and feedthrough the PWM from the NUCLEO PCB.
I have spent lots of time searching for grbl type code that includes backlash correction.  The only two I found was "GRBL-Advanced" and "Grbl Mega 2560 with Backlash Compensation".
Laird

J-Dunn, 2021-12-17:
Thanks, I thought you were intending to use the Nano to debug the issue. 
You should probably put any more comments on this in #75, not here. It's rather polluting the auto-squaring thread.