Schildkroet/GRBL-Advanced #57: Please use float point math

BalashovK, 2021-02-24:
I use atmel328p grbl with CNC and I had hardware damage because it cannot handle larger-radius arcs.
They did what they can - an engineering marvel of doing math with integers in such a limited space. Which kind of works but not suitable for working with expensive hardware (it cannot handle large values in I and J of G2, G3 commands and does not recognize it)

You are porting GRBL to a CPU with float point support. But you still keep all the integer math (in gcode.c)

Please, please refactor it to use float point math. Wherever you see word "mantissa" - replace it with float. Wherever you see multiplications and divisions, make sure parameters and result have type "float"

You will end up with industrial-grade board. Integer math is really the limiting factor of GRBL.
(another limiting factor for 328p GRBL is slow USB communication which makes fast laser engraving less achievable, but use of CPU with integrated USB resolves this problem)

Thank you for the great project!

BalashovK, 2021-02-24:
Here is an example G-Code (generated by Inkscape plugin "J Tech Photonics") that looks ok in G-Code viewers but makes GRBL drive motors to the bottom left way beyond the area:

M05 S0
G90
G21
;
G1 X-26.4249 Y49.0743 F2400
G3 X-26.4352 Y49.0642 I33030.0384 J-33718.6545 F2400
; 
M05 S0
G1 F2400
G1 X0 Y0

J-Dunn, 2021-04-04:
So obviously with 33030.0384 you overstepped the range of _int_
A quick fix would be to change everything to longint ( which the 32 bit chip will be doing with the ints anyway ). 

In principal a global search and replace operation but it will need some hand intervention to prevent accidental matches because "int" is so short and will likely appear may places other than type declarations.

Moving to fp may require a little more thought because of rounding errors and comparisons.

BalashovK, 2021-04-04:
J-Dunn, it is not that simple. 
G3 X-26.4352 Y49.0642 I20194.000 J0.0000 F2400 - works
G3 X-26.4352 Y49.0642 I20195.000 J0.0000 F2400 - triggers the error.
so yes, it is overflow, but not exactly at 32767 border.

In a meantime I wrote a converter for myself and it already saved by cutter from further damage a few times. I just replace G2 and G3 with G1 when I and J are large. 

I am not sure why switching from integer math to float point math would cause rounding errors.

J-Dunn, 2021-04-04:
The overflow is probably happening during the calculations not the final figure of the gcode. Why are your numbers so big?
Is that in thou' ?

BalashovK, 2021-04-04:
J-Dunn, yes, I would guess, overflow is occurring while calculating the path. 
Now, why numbers are so big. I was laser-cutting some design made in Inkscape and generated g-code with plugin "J Tech Photonics"
Perhaps, some of the curves were almost straight lines. The only curve supported by G-Code is an arc. So, somewhere in translation from beizer curves to arcs the plugin generated an arc with this enormous radius. Sadly, the plug-in does not have settings for maximum curve radius. And this the least buggy Inkscape g-code plugin I am aware of.

J-Dunn, 2021-04-04:
Yeah, I quickly gave up on Inkscape for Gcode.

BalashovK, 2021-04-04:
Well, Inkscape is still pretty powerful, and I use it a lot in my projects. If nothing else works, It can export to vector graphics formats like dfx.

Schildkroet, 2021-06-05:
This is not a problem with gcode/mantissa. All values are read correctly. The problem happens in the arc computation like @J-Dunn  said. But an arc with 33 meter radius is not a real life scenario

closed 2021-06-05 by Schildkroet

J-Dunn, 2021-06-05:
> Well, Inkscape is still pretty powerful, and I use it a lot in my projects. If nothing else works, It can export to vector graphics formats like dfx.

You may be able to import into FreeCAD, and clean up any silly small, nearly straight segements there,  but I would not say that is something I've had much success with. It's just a suggestion.