Schildkroet/GRBL-Advanced #59: BUG -Unable to change $0

Zorki4, 2021-03-17:
https://github.com/Schildkroet/GRBL-Advanced/blob/c64d012f96bce1a5c4ecedae686ebfc3da7e61d8/grbl/Settings.c#L400

Hi,
It is unable to change $0 by the command line, because function is commented out. There shouldnt be // marks.

`//settings.system_flags = int_value;`

Schildkroet, 2021-03-17:
This is intended since the pulse time (original $0) is fixed.
But i did some tests with internal flags, so i set them to $0, but thats not for normal users

closed 2021-03-17 by Schildkroet

Zorki4, 2021-03-17:
YES. I know that it is used fos inverting control pins (hold, start etc.) and I've tried to change it :) Befere I found that it is commented out I had to modify deafults.h, recompile and reprogram Nucleo board.