Schildkroet/GRBL-Advanced #61: warning: '<<' in boolean context

J-Dunn, 2021-04-04:
Hi,  I'm running a heavily modded  stm32 port of grbl 0.8 or so.  But it is built using A6 eclipse tools which is a lot of unnecessary dependency and eclipse is PITA wrt updates and constant need to reinstall stuff and PAINFULLY  slow in debugging mode.

I've wanted to find a Makefile solution for quite a while and some of the newer features like jog would be nice. So this project is great. It looks like it would be worth trying to integrate my mods across to get all these benefits.

I've just tried building for the first time and I get a couple of warnings. 

```
GRBL-Advanced/HAL/FLASH/eeprom.c:47:24: warning: '<<' in boolean context, did you mean '<' ? [-Wint-in-bool-context]
   47 |   checksum = (checksum << 1) || (checksum >> 7);

```

I see that's part of the HAL code, so I guess it's pretty well tested. Is it safe to ignore?

I'm using my distro's  arm toolchain:   gcc version 9.2.0

Thanks.

Schildkroet, 2021-04-04:
This is a known bug in the original grbl
https://github.com/gnea/grbl/issues/355

But i fixed this already last year. Are you using an old version?

J-Dunn, 2021-04-04:
ah, it looks like I did the checkout last Nov and never updated because I never got into it before.
Thanks for link. I'll do a git pull and try again.

J-Dunn, 2021-04-04:
Yes, builds perfectly thanks.

A little point on your readme it says Sungeun K. Jeon is the author.  He has been maintaining for the last ten years but he is not the author. That's Simen Svale Skogsrud. It would be nice to give credit to the right person :)

https://github.com/grbl/grbl

closed 2021-04-04 by Schildkroet