Schildkroet/GRBL-Advanced #62: please see issue 33, STM32F411CEU6 has 512 KB Flash and WeAct bootloader, but after .hex load there is no connection and com port

hunabapr, 2021-04-14:
I get the same issue, as is repoted in 33,  ie "STM32F411CEU6 has 512 KB Flash and WeAct bootloader, but after .hex load there is no connection and com port"

I've followed this up to 3 Apr 2020 and was getting the same results including the connect to the test prog.
But when I make the changes re:HAL/STM32/stm32f4xx.h plus... I still can't get this to work

Any ideas please?

Taken from ST-Link utility
14:31:15 : Device flash Size : 512KBytes
14:31:15 : Device family :STM32F411xC/E

The best Paul

hitchhiker85, 2021-09-13:
hi have you done it?

i am looking for the same port...
what about the portC pins what do you did ?

hunabapr, 2021-09-13:
No sorry, couldn't sort so moved on.

Schildkroet, 2021-09-22:
Sorry, i can't help you. I have no F411 available

closed 2021-09-22 by Schildkroet