Schildkroet/GRBL-Advanced #64: Makefile does not check .h files

J-Dunn, 2021-05-23:
The Makefile does not account for changes in any .h files outside of ARM. 
```

touch grbl/Config.h
bash-5.1$ nano Makefile
bash-5.1$ make
make[1]: '/svn/GRBL-Advanced/GRBL_Advanced.elf' is up to date.
make[1]: '/svn/GRBL-Advanced/GRBL_Advanced.bin' is up to date.
make[1]: '/svn/GRBL-Advanced/GRBL_Advanced.hex' is up to date.
make[1]: '/svn/GRBL-Advanced/GRBL_Advanced.lst' is up to date.
   text    data     bss     dec     hex filename
  66868     492   12164   79524   136a4 /svn/GRBL-Advanced/GRBL_Advanced.elf

```

J-Dunn, 2021-05-24:
I've been digging into how to do this and it is incredibly arcane and esoteric to achieve such a simple and bleedingly obvious requirement. 

In fact it seems that this boiler plate was intended to produce dependency file and include them but the dependency rule for .d files is missing. 

```
DEPENDS     :=  $(OFILES:.o=.d)
......
-include $(DEPENDS)

```
It seems that the relevant incantation and  hieroglyphics are provided here:

http://www.gnu.org/software/make/manual/make.html#Automatic-Prerequisites

Schildkroet, 2021-09-22:
I know this problem exists, but i'm far from an expert in makefiles. And it only takes 2-3s for a complete rebuild, so it is not a real problem for me.
If you have a feasable solution, you can ofc tell me

closed 2021-09-22 by Schildkroet