Schildkroet/GRBL-Advanced #65: Inconsistent "OEM" instructions in Config.h

J-Dunn, 2021-05-23:
Config.h  :
```

// If doing so, simply comment out these two defines and see instructions below.
#define DEFAULTS_GENERIC

// Serial baud rate
#define BAUD_RATE   115200
//#define BAUD_RATE 230400

```
There are not "two defines" , just one. 
```

   Instructions: Paste the cpu_map and default setting definitions below without an enclosing
   #ifdef. Comment out the CPU_MAP_xxx and DEFAULT_xxx defines at the top of this file, and
   the compiler will ignore the contents of defaults.h and cpu_map.h and use the definitions
   below.

```

There is no  CPU_MAP_xxx define. 

grbl/defaults.c  : 
```

#ifdef DEFAULTS_GENERIC
    // Grbl generic default settings. Should work across different machines.

```

It seems this has been changed and instructions are no longer consistent.

Schildkroet, 2021-05-23:
Currently there's no centralised point for cpu mapping. You would need a full abstraction layer for this feature. For now it is necessary to modify all involved files.

J-Dunn, 2021-05-23:
Thanks. I was not meaning a full HAL, just that all these settings were in the same file like they were on GRBL. Then you just needed to go to one place to see all the values that were needed and what pins they are configured to. 

On my original GRBL 0.9x this was all in a file cpu_map.h

J-Dunn, 2021-05-25:
I define these utility functions in nut_bolt.c to help clean up the code on stm32 :

```

void set_as_output(GPIO_TypeDef* port, uint32_t pin)
{
  GPIO_InitTypeDef GPIO_InitStructure;

  GPIO_InitStructure.GPIO_Pin = pin;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_OUT;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_PP;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_25MHz;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL;
  GPIO_Init(port, &GPIO_InitStructure);
}

void set_as_input(GPIO_TypeDef* port, uint32_t pin)
{
  GPIO_InitTypeDef GPIO_InitStructure;

  GPIO_InitStructure.GPIO_Pin = pin;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IN;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_PP;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_25MHz;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;
  GPIO_Init(port, &GPIO_InitStructure);
}

void set_as_AF(GPIO_TypeDef* port, uint32_t pin)    // connect pins to alternate function
{
  GPIO_InitTypeDef GPIO_InitStructure;

  GPIO_InitStructure.GPIO_Pin = pin;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_PP;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_25MHz;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL;
  GPIO_Init(port, &GPIO_InitStructure);
}

```
then for example in stepper.c : 

`	set_as_output(STEP_EN);
`
grbl/cpu_map/cpu_map_stm32f411_nucleos.h:180:   
#define STEP_EN    GPIOA,GPIO_Pin_9   // D8

All the h'w specifics are pulled out to a central file for easy reference and don't clutter the readability of the functional code. 
Makes life a whole lot easier if you need to move a pin.

closed 2021-08-07 by Schildkroet