Schildkroet/GRBL-Advanced #66: One board test.

J-Dunn, 2021-05-31:
Hi, I just wanted to try testing this code so I set up a run one motor test , a F411RE Nucleo with X step/dir/enable wired to a NEMA23 via DM542 driver unit. 

I built the code and the flash looked good:

```
2021-05-31T17:35:44 INFO flash_loader.c: Clear DFSR
2021-05-31T17:35:44 INFO common.c: enabling 32-bit flash writes
2021-05-31T17:35:45 INFO common.c: Starting verification of write complete
2021-05-31T17:35:45 INFO common.c: Flash written and verified! jolly good!

```

I opened Arduino IDE to act as a serial interface with /dev/ttyACM0 

On connecting the USB to power up the F411RE , I get a couple of garbage characters, that's OK. 

Then I press Enter to send a blank line to GRBL and get "ok" back. Looks good. 

However, whatever else I type it seems to lock up. I don't get anything else back until I power cycle it. I sent $$ locked: $# locked: ? locked.

Now this is probably me being dumb since I have not worked on a stripped down rig like this for a long time, so can you tell me if I'm messing up. Thanks

Schildkroet, 2021-05-31:
Hello,
did you modify the nucleo board?
Did you update the st-link to the latest version?
Did you modify the source code / config?

J-Dunn, 2021-05-31:
Hi, 

board is used but not modified. ( other than prior flashing ) 
```
 st-info --probe
Found 1 stlink programmers
  version:    V2J22S5

```
I have not changed the firmware because IIRC latest version limits some used of the st-link as a h/w tool. 

I did a few changes in Config.h , like turning off backlash. I should probably try a virgin config until this is sussed. 

OK, just did a clean git clone , rebuilt and got same result.

Schildkroet, 2021-05-31:
Can you pls test this file
[GRBL_Advanced.hex.zip](https://github.com/Schildkroet/GRBL-Advanced/files/6572033/GRBL_Advanced.hex.zip)

(First startup takes several seconds to write settings)

J-Dunn, 2021-05-31:
Thanks,
st-flash write GRBL_Advanced.hex 0x8000000  ( cf Makefile is flashing .bin )

I'm not getting any serial comms with that. Not even the ok.  Though I can still interrogate the st-link.

Schildkroet, 2021-05-31:
Second try
[GRBL_Advanced.hex.zip](https://github.com/Schildkroet/GRBL-Advanced/files/6572084/GRBL_Advanced.hex.zip)

Did you wait atleast 10sec to write settings?

J-Dunn, 2021-05-31:
yep waited 20s just to be sure, after LED on stlink stops dancing.   
No comms. 

It stays green BTW. If I power cycle it come back red.

Schildkroet, 2021-05-31:
I dont have an F411 with me right now, will test it later this week

J-Dunn, 2021-05-31:
BTW git clone just got me "software" branch, is that what I should be working with ?

Schildkroet, 2021-05-31:
Yes

J-Dunn, 2021-06-01:
Thanks. This is the kind of output I'm getting. The ok's are from sending newline the others were from sending $$ and $#.

So the data is not getting garbled as happens with a baud mismatch but very few bytes sent are getting recd.

This also seems to imply that grbl is still running fine ( not locking up as I first suspected ), but that there is a coms problem.
```

ok
ok
ok
.001
$ok
ok
ok
ok
59:0.00ok
ok
ok
ok
ok

$22=1
00ok
$0=16
12=0.00
010
ok
ok
ok

```

J-Dunn, 2021-06-02:
OK, I just tested this on two other PCs and it's fine. This must be something to do with the  mobo/BIOS. 

[EDIT] After fiddling with a couple of BIOS options this no looks to be working properly. Phew !! 

It took me rather longer than it should to narrow down where this was coming from but I've got it sussed. 

I'll close this issue now, since it's not a software issue. 
Many thanks for all the help.

closed 2021-06-02 by J-Dunn