Schildkroet/GRBL-Advanced #67: Thanks for the drilling cycle you added for grbl, but...

hankecnc, 2021-06-05:
I am from China and I am a cnc fan. Thank you very much for your Candle. He is very good (I really hate the interface of MACH3, it is complicated and ugly), and thank you very much for the drilling cycle you added for grbl, (I know ug can The drilling cycle is converted to a movement composed of g1), but in the standard industrial cnc system, it is usually after a G83 command, and other identical G83s will be omitted, and only the X and Y coordinates will be written. If you send this code to grbl An error will be reported, ERROR: The G code command is missing a required value word. (This article was translated by Google. Since I am not a professional programmer like you, I have no ability to modify him.)
Sample code: G98 G83 X0. Y0. Z-20. R3. F300. Q2.1251
X4.739
X19.183 Y-3.87
X17.689 Y-7.477
X15.313 Y-10.574
X12.216 Y-12.95
X8.61 Y-14.444
X4.739 Y-14.953
X0.869 Y-14.444
X-2.737 Y-12.95
X-5.834 Y-10.574
[image]
[image]

Schildkroet, 2021-06-05:
Hello,
in grbl you have to specify the g-code in every line. Just sending positon values doesn't work.

hankecnc, 2021-06-06:
> 您好，
> 在 grbl 中，您必须在每一行中指定 g 代码。仅发送位置值是行不通的。

G1 X50.5
Y50.5
X49.5
Y49.5
X50.
Y48.
X52.
Y52.
This is not the case. Codes like this are very common in my work. The first line has been given G1, and the latter is omitted.

I used grbl here before, he can perfectly support such code, https://github.com/robomechs/6-AXIS-USBCNC-GRBL

hankecnc, 2021-06-06:
I want to run this version of yours for my machine, but this problem cannot be solved

Schildkroet, 2021-06-06:
This behaviour is not supported for drilling cycles. You have to send the whole command in absolute mode for every hole

hankecnc, 2021-06-06:
> 钻孔循环不支持此行为。您必须以绝对模式为每个孔发送整个命令

But even G1, G2, and G3 do not support this

hankecnc, 2021-06-06:
malfunction

https://user-images.githubusercontent.com/69781416/120924916-aa8d8800-c708-11eb-8ac9-2600ff0fa0c9.mp4

hankecnc, 2021-06-06:
This is grbl USBCNC, the same code, can run normally

https://user-images.githubusercontent.com/69781416/120924945-cee96480-c708-11eb-95f3-6d5c6ba625e0.mp4

Schildkroet, 2021-06-06:
Can you send the gcode file?

hankecnc, 2021-06-06:
[Uploading 螺旋与平面圆测试6mm铣刀.txt…]()

Schildkroet, 2021-06-06:
> Uploading 螺旋与平面圆测试6mm铣刀.txt…

> This behaviour is not supported for drilling cycles. You have to send the whole command in absolute mode for every hole

Like i said, you have to send G83 for every hole

closed 2021-06-06 by hankecnc

hankecnc, 2021-06-06:
> > 正在上传平面与平面圆测试6mm铣刀.txt...
> 
> > 钻孔循环不支持此行为。您必须以绝对模式为每个孔发送整个命令
> 
> 就像我说的，你必须为每个洞发送 G83

Thank you, I think I can preprocess the G code on the host computer to solve it.
You are a good person, and your code is very clean and beautiful.