Schildkroet/GRBL-Advanced #69: When the output pins of the stepper motor driver are changed, it gives a fixed output from some of its old pins.

Abdssmtzl, 2021-06-29:
Hi, 

      First of all, thank you for sharing such an excellent project. I wanted to make a few changes to add a touchscreen to your project but there were some issues with pin changes.
      I was able to adapt your source file to STM32F429ZIT Discovery. However, in order to use the touch screen, I need to change the location of the stepper motor driver pins and limits. To begin with, I changed the stepper motor pins. There was no problem in its operation, but even though the old stepper motor pins were canceled, I get constant high outputs after installing the program from some of them.

I made the pin change only from GPIO.h files.  

Pins with high output: PA8, A10, B4
The new output pins are in order(stepX , DIRX , Sy, Dy, Sz, Dz); D11,D4,D5,D7,D2,D13
The new limit pins are in order:   E4,E5,F6,G2,G9,G3 (x1,x2,y1,y2,z1,z2) 

      Finally, when changing the limits, is there anything I should pay attention to when choosing a different empty timer pin instead of TIM4(B6)?

              Thank you for your interest.

Schildkroet, 2021-07-11:
Hello,
in GPIO.c and TIM.c the pins are only initialized. You have to search the code to see where they are used. There is no full abstraction layer available yet.

Abdssmtzl, 2021-07-12:
Change 1:      STM32F429ZIT Discovery KIT used.
Problem 1:    When pressing the reset pin, 3 pins give high output.

Change 2:     While new stepper motor driver input pins were defined, old pins were deleted from GPIO.c and GPIO.h files.
Problem 2:   3 of the deleted definitions give a constant high output. There are no problems with the new pins.

**Solution 2 :**  I think that deleted pins may be defined as inputs in the processor. Deleted pin definitions are defined in GPIO.c file. No addition was made to the GPIO.h file. It no longer outputs high. But when the reset pin is high, the old 3 stepper driver pins give high output. There are no problems with the new pins for now.

Change 3 :    There was no problem in changing the limit pins.

Pins with high output: PA8, A10, B4

Schildkroet, 2021-07-12:
Unused pins are floating input by default
But since you modified the source i dont know your problems

closed 2021-08-07 by Schildkroet