Schildkroet/GRBL-Advanced #70: Clarification on encoder

Steph101, 2021-07-08:
Does the system use a single pulse or AB encoder for threading feedback and which pins are used for this?

ClockedLogic, 2021-07-11:
Looking at the code it appears to count pulses per rev which indicates AB phases for proper quadrature.  In fact only A and index are required although the code doesn't mention the pins connected it uses tim2 for the encoder but there is no initialisation code.  You would need one pin for phase A or B input and a second pin for Index to reset the counter.   
TIM4 init code contains the following comment. 

  //Remap Encoder pins to PB8 and PB9. Previosly on 6 an7 where I2C1 should be

Schildkroet, 2021-07-11:
Hello,
currently i'm using an AB encoder. It is connectecd to PB 6/7
https://github.com/Schildkroet/GRBL-Advanced/blob/software/HAL/TIM/TIM.c#L136-L183

But it is still experimental

closed 2021-08-07 by Schildkroet