Schildkroet/GRBL-Advanced #75: CNC spindle motor PWM signal goes off (zero speed) when Backlash compensation makes a correction.

lairdgit, 2021-08-25:
I downloaded this version source:
#define GRBL_VERSION            "1.1i"
Revision 2
Schildkroet released this on Apr 15, 2020
---
I compiled it with EmBlitz 1.11.
I uploaded the code to a STM32 Nucleo F411RE using STM32 ST-LINK Utility v4.6.0.0

I have a garden variety CNC machine.
Using a dial indicator the backlash correction works great.

However, when using the CNC to mill some stock, the PWM signal goes to the inactive state when the backlash correction makes a correction.  The spindle motor is always slowing down and speeding up.

This may be great for a laser machine but not for a spindle motor.
Laird

Schildkroet, 2021-12-16:
I was busy this year and hadn't a lot of time. But i will definitely look into this issue the next weeks

Schildkroet, 2021-12-16:
Can you test if this works for you? @lairdgit

lairdgit, 2021-12-17:
> Can you test if this works for you? @lairdgit

YES your fix works.  Wonderful, just wonderful.
Thank you very much for your work and the time it takes.
Thanks again, Laird 😁

closed 2021-12-17 by Schildkroet