Schildkroet/GRBL-Advanced #80: CoreXY mode Not working

trkadam58, 2021-12-19:
When I have changed to CoreXY in  config.c file to #define COREXY. Firmware not working with software.

Schildkroet, 2021-12-19:
Can you please explain more in detail what is not working?

trkadam58, 2021-12-20:
- Please check below attached images,  i have changed  some line for COREXY and compiled, while compiling there is not have any error but Grbl advance not supporting with STM32F466RE. 

[image]
[image]

Schildkroet, 2021-12-20:
Please try the current commit

trkadam58, 2021-12-21:
I have taken trial with new code it is working fine,
Thanks

closed 2021-12-21 by trkadam58

trkadam58, 2021-12-22:
Today I have taken trial with laser cutting software
 LIGHT BURN software, it is working perfectly,

trkadam58, 2021-12-22:
Good morning, 
 I am very happy with your coding, so i want to make some small donation, but i not find the any link, please share payment link to trkadam58@gmail.com.

Thanking you,
Tukaram