Schildkroet/GRBL-Advanced #83: Laser support?

Edocecrous, 2022-02-21:
Hi,
I couldn't find any means to contact you, i hope i'm not intruding by placing this under 'issues', even though it's more like a feature request.
I'm replacing my Arduino UNO + cnc shield in my laser with Arduino MEGA + Ramps 1.4, but i would be much happier with GRBL-Advanced and an STM board. And was wondering if there is a possibility to include the 'laser' feature of the Arduino GRBL version.

Schildkroet, 2022-02-21:
Hello,
what exactly is your problem?
Several people used it successfully for laser engraving and helped finding bugs.
https://github.com/Schildkroet/GRBL-Advanced/issues/3
https://github.com/Schildkroet/GRBL-Advanced/issues/4
https://github.com/Schildkroet/GRBL-Advanced/issues/35

Edocecrous, 2022-02-21:
In that case, my sincere **apologies**, based on what i red so far it seemed the laser functionality is not implemented.
I have an STM32F411CEU6, what i would like to use to control my K40.
If i understand correctly, then i can send grbl a `$32=1` command and turn on laser mode?

Schildkroet, 2022-02-21:
Laser should work just like in original grbl

closed 2022-02-21 by Schildkroet