Schildkroet/GRBL-Advanced #91: Delay Movement when backlash compensation

hanifhan, 2024-09-25:
hi , ive been trying to use a backlash compensation , but i found out there is kind of delay ( accel/deccel ) movement when the machine do a compensation backlash , so the movement became not smooth , is it designed like that or any explanation on it , because it will impact in machining time  ?

machining time when use backlash compensation : 13 minutes and 40 second
machining time when not using backlash compensation : 10 minutes  and 20 second 

https://github.com/user-attachments/assets/fd6b0721-ad47-4418-af29-57f8b1a70e2c

hanifhan, 2024-09-26:
yea , i update the lastest grbl advanced
here is my $$ update and the gcode

$0 = 0    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.001    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$14 = 0   
$15 = 360   
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 3000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 0   
$34 = 1   
$35 = 0   
$36 = 1   
$37 = 0   
$38 = 0   
$39 = 1   
$40 = 1   
$41 = 0   
$42 = 0   
$100 = 500.000    (X-axis travel resolution, step/mm)
$101 = 500.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 12500.000    (X-axis maximum rate, mm/min)
$111 = 12500.000    (Y-axis maximum rate, mm/min)
$112 = 12500.000    (Z-axis maximum rate, mm/min)
$120 = 400.000    (X-axis acceleration, mm/sec^2)
$121 = 400.000    (Y-axis acceleration, mm/sec^2)
$122 = 400.000    (Z-axis acceleration, mm/sec^2)
$130 = 400.000    (X-axis maximum travel, millimeters)
$131 = 400.000    (Y-axis maximum travel, millimeters)
$132 = 500.000    (Z-axis maximum travel, millimeters)
$140 = 0.010   
$141 = 0.010   
$142 = 0.010
[ALL.txt](https://github.com/user-attachments/files/17141633/ALL.txt)

Schildkroet, 2024-09-26:
You need to increase axis-acceleration if possible, so the backlash move can be performed faster.

hanifhan, 2024-09-27:
may i knew what is the cause of it ? maybe i could try to fix it ?

Schildkroet, 2024-09-28:
This is a rather complicated physics issue.
While one axis is moving with F7000 in your gcode, the backlash axis is changing direction (no movement). In theory, the backlash move has to be done in 0 time, so it does not affect any position, but thats just not possible with a "slow" mechanical system. The time it takes to do the backlash move depends mostly on acceleration and distance of backlash. With high acceleration and small distance, the backlash move can be done in small time and with fast speed. Otherwise the other axis has to slow down.
If it would be that easy to fix, i and many others would have already done it.

closed 2024-09-28 by Schildkroet